meshtastic/home-assistant #12: Send to channel

asjmcguire, 2024-12-22:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I use the MQTT JSON feature at the moment to broadcast weather data on 2 channels, primary for less frequent weather updates, and road conditions when the temperature drops below 3C. And a dedicated channel for broadcasting the current conditions every 30 minutes. There only appears to be the ability to broadcast to another Node at the present moment, in this integration.

### Describe the solution you'd like

The ability to broadcast to any configured channel for the selected node.

### Describe alternatives you've considered

I will have to continue using MQTT for the time being.

### Additional context

Requires ability to broadcast to channel index. Eg 0 or 2 in my case.

broglep, 2024-12-22:
Thanks for reporting and helping prioritize different features. There are already entities for the individual channels, but so far there is now way via home assistant to send to a channel. 

One idea is to offer the channels also as [notify target](https://developers.home-assistant.io/blog/2024/04/10/new-notify-entity-platform/) (same as all nodes). Another option is to extend send_text service or introduce dedicated service.
Unfortunately device actions don't work that well as channels are not devices themselves.

Do you have a preferred way of integrating that functionality?

asjmcguire, 2024-12-22:
> Do you have a preferred way of integrating that functionality?

So from a GUI standpoint, I thought it would make the most sense for the channels you have already configured on the node, to appear in the dropdown for the target.

However, as this may be more complicated for you to implement, in the short term at least, I'd be more than happy to switch to YAML and use something like ```channel_index: 0``` to indicate that I intend this to be a broadcast message rather than a direct to node message.

broglep, 2024-12-22:
> So from a GUI standpoint, I thought it would make the most sense for the channels you have already configured on the node, to appear in the dropdown for the target.

Are you referring to automation UI where you select currently the device as target or are talking about the Developer Tools -> Actions (meshtastic.send_text service)?
For services/actions it is easy doable to offer the entities as target (the new notification system also works that way)

asjmcguire, 2024-12-22:
The "to" field on the automation editor after you have selected the node that you are sending the message from.
I guess an easy way to do it on the GUI would be to have a tickbox for the node you intend to send the message to, and another field with a tickbox that is just to type in a channel number. Then you can easily just type the channel number, rather than having to have a dropdown of the channels, as I assume if you have multiple nodes added, it will make more complicated since taking my example - borderswx is channel 2 on the main node, but might be channel 1 on another node. So it's probably easier to allow someone to just directly state the id of the channel they want to broadcast from?

broglep, 2024-12-22:
Ok, thanks for the clarification, so you are directly using the action/service in your automation, and not device action. 

The send_text action actually supports not only home assistant device id as `to` value, but also node number or the user id, but that is not really exposed via UI and UI does not offer that much flexibility. Doing similar with with channel will be possible as well.

I had to change the structure of some internal events to be more explicit for the current feature I'm working on to store direct messages and channel messages, so I might end up introducing similar change for the send text service as well, this would exactly be as you describe it, separate fields for node or a channel as target (and probably supporting both entity / device ids as well as numeric ids)

Do you refer to the channel index (node local concept) or the channel hash (mesh protocol concept)) when you are suggesting to use as it as identifier? 

You are right that the channel a tricky things as the channel numbers don't necessary have to match on different devices. The only real unique identifier for a channel is its PSK, so channel index is no good way to identify a channel when you are not at the same time exactly specifying via which gateway node you are sending a message. The channel hash might work as short version of the PSK that could be used by a human, but that will be probably confusing as well as you can't really distinguish a channel index from a channel hash as a human.

I think the safest way will be that via UI you are only able to select channel entity, such an entity belongs to a device and you can be sure that it is sending on the correct channel. This pre-determines the gateway node you are sending from, but that shouldn't be such a big issue. 

Due that complications with channels, it might not be well suited for the notify platform, as this integration's concept for notify targets is that they are independent of the gateway, but maybe with notify entities there would be a way to use PSK or channel hash as the identifier

Depending on how much the use cases will differ in the end between private message and  channel message I might end up introducing to separate action/services.

asjmcguire, 2024-12-22:
> I'm working on to store direct messages and channel messages,

Ooooh yay!

> Do you refer to the channel index (node local concept) or the channel hash (mesh protocol concept)) when you are suggesting to use as it as identifier?

I don't know how the Meshtastic Python API works for this part, so that's really up to you to decide which is the easier method, from the human point of view though, simply using the channel index makes the most sense. Eg 0 - 7

I provide a screenshot of my current dashboard which I populate via MQTT - so you can see the things I am most interested in.

[image]

broglep, 2024-12-23:
Nice dashboard! As far as I can tell the packet types metrics are the only ones that are currently not available (although there are some slightly related metrics for the gateway: packets received, relayed, sent, duplicate, etc)

I've just released new version 0.3.0 that comes with quite some improvements, including support to send to channel. 

I have kept `meshtastic.send_text` service/action for the time being and added option to specify channel as well. It is still used internally for certain use cases, but I would not recommend using it in automation (I don't consider it as public interface and there could be breaking changes in the future). For your use case of sending message to a channel, I'd recommend to use the new `meshtastic.broadcast_channel_message` action or use the `notify.send_message` with channel notify entity (the later maybe bit less final, unique id / entity ids might still change, but so far looks pretty good)

asjmcguire, 2024-12-23:
I will updated tomorrow - during daylight. My Home Assistant gets a bit upset when I update at night, because the Solar inverter powers down when there is no solar, and so all my solar related entities stay as Unavailable. I'll give the update a try and let you know how I get on with it. Thank you. I hope you got the donation earlier.

asjmcguire, 2024-12-25:
Getting a lot of 
```
2024-12-24 23:45:00.060 ERROR (MainThread) [homeassistant.components.automation.meshtastic_weather_broadcast] Meshtastic Weather Broadcast: If at step 1: If at step 1: If at step 1: Error executing script. Error for call_service at pos 2: Not connected
2024-12-24 23:45:00.060 ERROR (MainThread) [homeassistant.components.automation.meshtastic_weather_broadcast] Meshtastic Weather Broadcast: If at step 1: If at step 1: Error executing script. Error for if at pos 1: Not connected
2024-12-24 23:45:00.060 ERROR (MainThread) [homeassistant.components.automation.meshtastic_weather_broadcast] Meshtastic Weather Broadcast: If at step 1: Error executing script. Error for if at pos 1: Not connected
2024-12-24 23:45:00.061 ERROR (MainThread) [homeassistant.components.automation.meshtastic_weather_broadcast] Meshtastic Weather Broadcast: Error executing script. Error for if at pos 1: Not connected
2024-12-24 23:45:00.061 ERROR (MainThread) [homeassistant.components.automation.meshtastic_weather_broadcast] Error while executing automation automation.meshtastic_weather_broadcast: Not connected
```
If I go in to the automation and manually run the action, it will immediately come back and say "Not Connected" for the first few times I try to run the action, and then finally will succeed.
I'm connected over IP right now, I tried Bluetooth too, but could never get it to connect even after removing the PIN, and then I tried serial, which does work, but stopped updating the data once the uptime reached 497 seconds, and additionally if I try to add one of the nodes it can see from the configure button (like I do when connected over WiFi - which works fine) - all entities become unavailable and it never adds the node. 

You might also like to know though - there were many log entries complaining about the platform not using unique id's - I will open a bug report about that.

broglep, 2024-12-25:
Did you use `notify.send_message` or `meshtastic.broadcast_channel_message`?

I already experienced similar thing when using `notify.send_message` with a channel and multiple config entries / gateways and few of them not working (but that should kind of remediated now as `notify.send_message` does try all gateways now).
But `meshtastic.broadcast_channel_message` does not yet support this, so when you are not specifying `from`, you can always end up with the same non-working gateway node to be used. 

You can try to enable debug logging for the integration? I occasionally see continuous reconnect loop with TCP connection that I can only recover manually by either restarting the integration or the meshtastic node, such a reconnect loop is visible in the debug logs, that could be another cause for that not connected error. As the initial connect & full read-out of all data always succeeds before the connection is dropped, it can appear on the surfice that still everything is working. 
Still have to figure out how to automatically recover in such a situation, one of the issues is that the meshtastic node does drop previous tcp connections if it gets a new one, so if something is stuck and for some reason multiple connections are open / in-flight, you end up in such a situation. 

Bluetooth and serial are indeed rather unstable compared to TCP based on my tests. Bluetooth stability highly depends on the dongle used and for serial it seems that sometimes the serial interface completely disappears after some time (suspect it to be a hardware / firmware issue, for that particular node bluetooth also stops completely and need to power cycle the device)

asjmcguire, 2024-12-25:
> Did you use `notify.send_message` or `meshtastic.broadcast_channel_message`?

Hi, sorry, I wasn't expecting a response over Christmas haha.
Yeah I'm using the ```meshtastic.broadcast_channel_message``` option.
I plan to tidy up my network a little bit to reduce the number of switches that traffic is passing through, in the hope that might help somewhat too. But I have to wait till the end of December for my Netgear Gigabit Switch PSU to be shipped to me.
If that doesn't help clear things up a bit, then I will enable debug logging and see what it happening.

The interesting thing about the serial interface, was that while everything stopped updating, it would still continue to send the channel broadcasts, so it was definitely still connected.

broglep, 2024-12-25:
> Hi, sorry, I wasn't expecting a response over Christmas haha.

Won't be able to do much testing with actual devices next few days, but can help out with troubleshooting from time to time :) My long term off-grid test-setup/installation is currently pretty short on power as the snowfall has cut both power sources, so won't have much long term testing data with the newest 0.3.0 version. 

> I plan to tidy up my network a little bit to reduce the number of switches that traffic is passing through, in the hope that might help somewhat too.

Not sure if the network itself plays an role with that reconnect loop issue, more suspect meshtastic firmware in combination with operation system / python + the integration behaviour. Maybe bigger backoff times for re-connects will help, but will have to investigate further

>  If that doesn't help clear things up a bit, then I will enable debug logging and see what it happening.

You can actually also see the reconnect loop on the meshtastic node side if you have a serial log console open for that node you are connecting via tcp. 

> The interesting thing about the serial interface, was that while everything stopped updating, it would still continue to send the channel broadcasts, so it was definitely still connected.

Just for clarification, you mean you could still send messages via the serial interface? 
In my case the the node is not able to communicate at all, but it still properly working on the mesh (receiving & relaying messages)
 
> Oh, and by the way I am running the Alpha firmware released today. 2.5.17.b4b2fd6

I'm also testing mostly with recent alpha / beta firmwares, ranges from 2.5.11 - 2.5.17

asjmcguire, 2024-12-25:
> Won't be able to do much testing with actual devices next few days, but can help out with troubleshooting from time to time :) My long term off-grid test-setup/installation is currently pretty short [...]

Oh gosh! That doesn't sound good. It's practically a heatwave in the UK right now. We were at 14.1C today, definitely no white Christmas here this year.

> Just for clarification, you mean you could still send messages via the serial interface?
> In my case the the node is not able to communicate at all, but it still properly working on the mesh (receiving & relaying messages)

When I used configure to try to add one of my other nodes that the gateway could see, when the gateway was connected via serial, and the node was not added, and all the gateway entities went unavailable - messages could not be sent. The serial was as far as I can tell fully disconnected at this point. As a reload of the integration would result in a message saying failed to connect.

When I ONLY added the gateway via serial, and did not attempt to add any of the other nodes that it could see, the data stopped updating after about 500 seconds (uptime didn't continue to increase, airtime remained static etc). But messages continued to send.

asjmcguire, 2024-12-29:
In the short term, until you get the connection issue figured out, I'm currently using a loop with a delay, to try and ensure the message sends, which improves things a bit, but still isn't terribly reliable. I'm wondering, can we have an event for acknowledgement received? Then I could use the wait for trigger action, and just keep retrying until I get an acknowledgement that it has been sent.

Otherwise I noticed that there are events for message sent, which I could potentially use, because I assume this won't fire until the message is actually definitely sent. But I really need a bit of documentation in order to be figure out what the event would be ```meshtastic.channel_message.received``` I assume, but what event data is returned with this?

broglep, 2024-12-30:
I have some improvements locally (exponential backoff delay for failed reconnect and "circuit breaker" if many reconnects happen in short succession) that I still have to test a bit further. 

Events for ack's won't work that well as you will have to match the ack to your message, and it might be tricky im home assistant. My goal so far was to give synchronous felling service calls and offer the ability to decide how long / what to wait for (wait for transmission, wait for first ack, wait for response) with flags. 

If you have `ack` enabled for the `meshtastic.broadcast_channel_message` service call, it only completes after an ack is received (or the timeout is triggered after 30 seconds), so you should be fine using just that in a loop.

I don't think the `meshtastic.channel_message.received` will work, they are triggered when the gateway receives a channel message and acks to messages are not sent/received as text message (they use routing portnum). Those events are used for device triggers and only contain device id & entity id and text message

asjmcguire, 2024-12-30:
I did some experimentation in the Events listener of dev tools last night, and established that ```meshtastic.channel_message.sent``` which is of course what I meant, rather than received - won't work for me, because ALL the service calls for ```meshtastic.broadcast_channel_message``` always fail. It either fails with "Not Connected" in which case the message is not sent, or it fails with "Connection interrupted" in which case the message is actually sent.

I don't know what it is about this specific service call, because I tested listening for api_node_update and telemetry and position requests and all that works without issues, it's only the sending message which seems to believe that the TCP connection is not connected?

I'm going to reflash the gateway again, but this time I am going to choose to erase it fully, rather than just flash the app partition, I will see if that makes any difference at all.

broglep, 2024-12-30:
I don't think that full reflashing will make a difference. Did you have the possibility to enable the debug logs for the integration in home assistant or look at the meshtastic node's serial log? It would good to know if it is the reconnect loop, or maybe even another issue that causes the connection to be interrupted. (e.g. some other components such as web interface that is opening tcp connection)

I have pushed my local changes to the dev branch, so if you want to give it a try, you can manually update/replace the current interface.py with https://github.com/broglep/homeassistant-meshtastic/blob/dev/custom_components/meshtastic/aiomeshtastic/interface.py

asjmcguire, 2024-12-30:
I got the Meshtastic CLI running on my laptop and I am indeed seeing a lot of this:

```
[0m [32m INFO  [0m| 15:17:26 50 [ApiServer] [32m Force close previous TCP connection
[0m [34m DEBUG [0m| 15:17:26 50 [ApiServer] [34m PhoneAPI::close()
[0m [32m INFO  [0m| 15:17:26 50 [ApiServer] [32m Incoming API connection
[0m [32m INFO  [0m| 15:17:26 50 [ApiServer] [32m Incoming wifi connection
```
Roughly every 8-10 seconds I would say. It's not clear why exactly the meshtastic firmware is force closing an in use ongoing TCP connection though? I don't have anything else active, the phone app is not connected to the device, the web interface is not in use. I don't understand why it keeps closing the connection.

broglep, 2024-12-30:
Ok, then it looks quite like what I see when I get into that reconnect loop. Can you try with the new interface.py? 

I still not fully understand it, it can be that with the connection the integration itself is creating can cause that situation. integration detects that the connection is closed, then it re connects, and during/after the reconnect is complete, the connection is closed again and integration detects this and re-connects once more

asjmcguire, 2024-12-30:
I have fully removed it from homeassistant, updated interface.py, and restarted Home Assistant.
The last 4 messages have successfully been broadcast, but additionally there has been another improvement. The 2 remote nodes which initially had entities before I went through the process of trying to change how I was connecting to the gateway, never had any entities when I finally reconnected back to TCP, even though I had fully removed the integration before adding it back. I don't know if it's the new interface.py file, or restarting Home Assistant or something else. But as well as feeling much more stable, I now have ALL the entities for the remote nodes.

asjmcguire, 2025-01-03:
As you are triaging - I am happy to consider this feature request closed for now. It is happily doing the job I wanted it to do.

broglep, 2025-01-03:
I'm closing it now. Further details for tcp connection related problems will be tracked in #24

closed 2025-01-03 by broglep