meshtastic/home-assistant #17: Error Log: Platform meshtastic does not generate unique IDs.

asjmcguire, 2024-12-25:
### System Health details

## System Information

version | core-2024.12.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1502
Downloaded Repositories | 40

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 20 November 2025 at 00:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | eu-central-1-11.ui.nabu.casa
certificate_status | ready
instance_id | d74dbc9c9b6b4563b25a3c274331fe7a
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 14.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
[... 97 lines cut ...]
2024-12-23 11:42:34.393 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663091472_node_long_name is already used by sensor.meshtastic_3663091472_node_long_name - ignoring sensor.meshtastic_⛅_3663091472_node_long_name
2024-12-23 11:42:34.393 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663101144_node_long_name is already used by sensor.meshtastic_3663101144_node_long_name - ignoring sensor.meshtastic_⛅_3663101144_node_long_name
2024-12-23 11:42:34.393 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3662959096_device_uptime is already used by sensor.meshtastic_3662959096_device_uptime - ignoring sensor.meshtastic_⛅_3662959096_device_uptime
2024-12-23 11:42:34.393 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663091472_device_uptime is already used by sensor.meshtastic_3663091472_device_uptime - ignoring sensor.meshtastic_⛅_3663091472_device_uptime
2024-12-23 11:42:34.394 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663101144_device_uptime is already used by sensor.meshtastic_3663101144_device_uptime - ignoring sensor.meshtastic_⛅_3663101144_device_uptime
2024-12-23 11:42:34.394 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3662959096_device_battery_level is already used by sensor.meshtastic_3662959096_device_battery_level - ignoring sensor.meshtastic_⛅_3662959096_device_battery_level
2024-12-23 11:42:34.394 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663091472_device_battery_level is already used by sensor.meshtastic_3663091472_device_battery_level - ignoring sensor.meshtastic_⛅_3663091472_device_battery_level
2024-12-23 11:42:34.394 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663101144_device_battery_level is already used by sensor.meshtastic_3663101144_device_battery_level - ignoring sensor.meshtastic_⛅_3663101144_device_battery_level
2024-12-23 11:42:34.394 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3662959096_device_voltage is already used by sensor.meshtastic_3662959096_device_voltage - ignoring sensor.meshtastic_⛅_3662959096_device_voltage
2024-12-23 11:42:34.394 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663091472_device_voltage is already used by sensor.meshtastic_3663091472_device_voltage - ignoring sensor.meshtastic_⛅_3663091472_device_voltage
2024-12-23 11:42:34.394 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663101144_device_voltage is already used by sensor.meshtastic_3663101144_device_voltage - ignoring sensor.meshtastic_⛅_3663101144_device_voltage
2024-12-23 11:42:34.394 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3662959096_device_channel_utilization is already used by sensor.meshtastic_3662959096_device_channel_utilization - ignoring sensor.meshtastic_⛅_3662959096_device_channel_utilization
2024-12-23 11:42:34.395 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663091472_device_channel_utilization is already used by sensor.meshtastic_3663091472_device_channel_utilization - ignoring sensor.meshtastic_⛅_3663091472_device_channel_utilization
2024-12-23 11:42:34.395 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663101144_device_channel_utilization is already used by sensor.meshtastic_3663101144_device_channel_utilization - ignoring sensor.meshtastic_⛅_3663101144_device_channel_utilization
2024-12-23 11:42:34.395 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3662959096_device_airtime is already used by sensor.meshtastic_3662959096_device_airtime - ignoring sensor.meshtastic_⛅_3662959096_device_airtime
2024-12-23 11:42:34.395 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663091472_device_airtime is already used by sensor.meshtastic_3663091472_device_airtime - ignoring sensor.meshtastic_⛅_3663091472_device_airtime
2024-12-23 11:42:34.395 ERROR (MainThread) [homeassistant.components.sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_sensor_3663101144_device_airtime is already used by sensor.meshtastic_3663101144_device_airtime - ignoring sensor.meshtastic_⛅_3663101144_device_airtime
2024-12-23 11:42:34.396 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_binary_sensor_3662959096_device_powered is already used by binary_sensor.meshtastic_3662959096_device_powered - ignoring binary_sensor.meshtastic_⛅_3662959096_device_powered
2024-12-23 11:42:34.396 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_binary_sensor_3663091472_device_powered is already used by binary_sensor.meshtastic_3663091472_device_powered - ignoring binary_sensor.meshtastic_⛅_3663091472_device_powered
2024-12-23 11:42:34.396 ERROR (MainThread) [homeassistant.components.binary_sensor] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_binary_sensor_3663101144_device_powered is already used by binary_sensor.meshtastic_3663101144_device_powered - ignoring binary_sensor.meshtastic_⛅_3663101144_device_powered
2024-12-23 11:42:34.397 ERROR (MainThread) [homeassistant.components.device_tracker] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_device_tracker_3662959096_node_position is already used by device_tracker.meshtastic_3662959096_node_position - ignoring device_tracker.meshtastic_⛅_3662959096_node_position
2024-12-23 11:42:34.397 ERROR (MainThread) [homeassistant.components.device_tracker] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_device_tracker_3663091472_node_position is already used by device_tracker.meshtastic_3663091472_node_position - ignoring device_tracker.meshtastic_⛅_3663091472_node_position
2024-12-23 11:42:34.398 ERROR (MainThread) [homeassistant.components.device_tracker] Platform meshtastic does not generate unique IDs. ID 01JFR7ZBZXTW266P45XX2M4CS3_device_tracker_3663101144_node_position is already used by device_tracker.meshtastic_3663101144_node_position - ignoring device_tracker.meshtastic_⛅_3663101144_node_position
```
```

broglep, 2024-12-25:
Thank for reporting. It is kind of a known issue and can happen in certain circumstances, it can also happen for notify entities. I think in your case the issue is the old entities still lying around. 

You probably can workaround that issue when you reconfigure the config entry. You don't need to delete and re-add a gateway when the IP address changes. For the entry in question, use the three dots next to configure and select reconfigure.

broglep, 2024-12-28:
Another new instance when this can happen: If the gateway node loses its name, we fallback to default short name, and thus get different entity ids. Will also happen when short name of the gateway node is changed

`01JEVQ89HNDB8YYM4AAQJA91TY_sensor_627830047_device_airtime is already used by sensor.meshtastic_brig_627830047_device_airtime - ignoring sensor.meshtastic_e668_627830047_device_airtime`

asjmcguire, 2024-12-28:
I did notice that the shortname is being used in the entityid, and in my case that shortname is an emoji (which is encouraged). But I don't think we should be using an emoji in entityids inside homeassistant? It feels like the nodeid should be enough, and that generally speaking doesn't change because it's tied to the BT MAC of the device.

broglep, 2024-12-28:
Did notice that as well in your logs and was surprised. But nice that it actually seems to work perfectly fine in home assistant. I think there is nothing that forbids emojis in entity ids (probably the allow all unicode except some reserved characters)

Using short name was an attempt to get some more meaningful entity ids. At least the gateway is expected to have rather static short name. Otherwise you end up with `sensor.meshtastic_3771721320_627830047_device_airtime`, not that easy to work for humans, in particular if you import a node via two different gateways. A user can still rename the entity ids to their liking, but that is probably not that feasible with that many entities this integration produces

It is still unclear to me what the best strategy is for entity ids.

timniklas, 2025-01-03:
Im am having the same issue, the log is full of this errors

broglep, 2025-01-03:
> Im am having the same issue, the log is full of this errors

I suspect in your case it is caused by adding the same gateway via two different interfaces (TCP and serial)

timniklas, 2025-01-03:
> > Im am having the same issue, the log is full of this errors
> 
> I suspect in your case it is caused by adding the same gateway via two different interfaces (TCP and serial)

Im only using one TCP connection, another node arrives tomorrow for the (hopefully) more stabile serial connection.

djoris, 2025-01-14:
I get the same messages in my debug log, and a whole bunch of other issues. 
Is there a way to reset/remove all previous configuration? I tried removing and re-downloading via HACS, but previous config seems to remain. 
In the directory /config/custom_components/meshtastic I can't find a YAML or comparable to modify,

broglep, 2025-01-15:
Can you share some of the log message so that we know what instance of this problem yo have?

I don't know exact details when and how home assistant cleans old data. You will for need to remove the configuration in Home assistant and then restart. Removing the integration beforehand might prevent from cleaning up the old data.
If this does not help, you can remove the old data in /config/.storage/core.entity_registry while home assistant is not running and then start again.

agoode, 2025-04-28:
> Did notice that as well in your logs and was surprised. But nice that it actually seems to work perfectly fine in home assistant. I think there is nothing that forbids emojis in entity ids (probably [...]

I think the issue is that you are changing the name of the entity. It looks like the entity name starts out without the shortname, then when that is discovered, the entity is renamed to include the shortname.

I am guessing that entity names are not expected to change from the integration, only by the user in the UI.

This is probably also going to be an issue when nodes are renamed in the mesh. If you have automations set up, you're going to need to have a stable entity name, so it might be best to stick with a more opaque name.

iot-pro, 2026-10-18:
Hi all,

we've been hitting this on a current setup as well (2026-10-18):

4x per HA restart, integration v0.6.1, single TCP gateway (RAK4631, firmware 2.7.x-beta), HA Core 2026.10.1. One of our channels ends up unaddressable via  because the duplicate entity is silently dropped.

**Observation on the likely cause**

Looking at the entity IDs in our setup, the primary/default channel notify entity and the per-channel notify entity seem to collide on the same  unique_id base.

**Suggested fix**

Make the notify unique_id include the entity kind and channel index, e.g. , instead of only the channel hash. That keeps existing entity IDs stable for other users while removing the collision.

Happy to provide full log lines or diagnostics if that helps. Thanks for the great integration!

Thomas

jamesarich, 2026-11-09:
Two causes are fixed on main: coordinator updates re-added every entity because a new entity's `entity_id` is `None` (#244 matches on `unique_id`), and gateway, channel and DM entities now come from one config-entry platform (#237). A restart on 2026.10 logs no unique ID errors. Reopen if 0.7.0 still shows it.

closed 2026-11-09 by jamesarich