meshtastic/home-assistant #19: Not all messages are visible comparing to MQTT

leshniak, 2024-12-30:
### System Health details

## System Information

version | core-2024.12.5
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.1.21-v8+
arch | aarch64
timezone | Europe/Warsaw
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4988
Installed Version | 2.0.1
Stage | running
Available Repositories | 1497
Downloaded Repositories | 24

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>GIOŚ</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 13
views | 9
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 18, 2024 at 10:38
-- | --
current_recorder_run | December 24, 2024 at 01:14
estimated_db_size | 284.36 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Not all messages are visible comparing to MQTT. See below:

[image]

### Reproduction steps

Configure Meshtastic node via MQTT and this integration, read messages from the channels.

### Debug logs

```text
(I'll try to paste it in the following posts, because it freezes my browser for some reason)
```

broglep, 2025-01-01:
That is a nice test setup, will try to do something similar (although I'm not really close to place where lot of messages are sent)

Would you mind trying the dev version as per instructions in https://github.com/broglep/homeassistant-meshtastic/issues/12#issuecomment-2565483930 ? 
That is currently the only reason I can imagine why you would not receive all messages

broglep, 2025-01-02:
@leshniak Can you please try with the new released version 0.4.0 and see if you are still seeing missing messages?

leshniak, 2025-01-03:
Sorry for lack of response, I’m sick with the flu. 

I’ve just installed it and I’ll try to give you a feedback after some activity on channel.

broglep, 2025-01-03:
> Sorry for lack of response, I’m sick with the flu.
> I’ve just installed it and I’ll try to give you a feedback after some activity on channel.

No worries, get well soon. 
I did not yet find time to create a test setup similar to yours. A stress test would be quite interesting. The meshtastic range test module could probably be used to create some synthetic load and constantly send out message and then compare between MQTT and this integration recorded messages

leshniak, 2025-01-09:
Unfortunately some messages are lost on v0.4.0 too. Other readings are fine and stable.
[image]

broglep, 2025-01-12:
What would be helpful are debug logs of the home assistant integration as well as serial logs from the meshtastic node if that is possible to collect. If other readings are stable and fine, it seems strange the the messages themselves are not received all. 
One thing that would be interesting to see if the meshtastic firmware actually publishes the message as phone notification (the log message reads like `Tell client we have new packets`).
What meshtastic firmware version are you using?

leshniak, 2025-01-16:
Hey @broglep, here are some debug logs from HA. As you can see, the last message from 21:02 is missing. I'll try to collect logs from the device, but it's more tricky. I'm using v2.5.18.89ebafc firmware.

[image]

[log.txt](https://github.com/user-attachments/files/18445269/log.txt)

broglep, 2025-01-16:
Thanks for providing the logs. I could find the corresponding debug log for the text message on `19:37:57`, but none for the one that is missing at `21:02`, which is really odd as there are quite some packets being processed at that time (position and telemetry), so the connection itself seems to be properly functioning. 

Serial log outputs would be really helpful in this case, it almost appears like the text message is not being sent to the phone api. If I remember correctly mqtt publishing is done before sending notification to the phone api / client / this integration, some maybe something is preventing the notification to be sent on the firmware

leshniak, 2025-01-16:
I’m gonna setup second node and collect logs via serial, I will also setup rsyslog for the first, remote device. Hopefully we’ll find something.

leshniak, 2025-01-24:
I didn't have an opportunity to make this setup, I'll do this after February 6th.

leshniak, 2025-04-07:
Looks fine on v0.6 and firmware v2.6. Closing for now.

closed 2025-04-07 by leshniak