meshtastic/home-assistant #21: FEAT: Fire an event when a NEW node is found on the mesh.

asjmcguire, 2024-12-30:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I would quite like an event to be fired on the Home Assistant Event Bus when a new node is discovered on the mesh.

### Describe the solution you'd like

So the use case would something like this:
people drive through the village I live in, if there is a Met Office warning (UK) in force, upon a new node announcing itself to the mesh, I would then automatically send that node a direct message warning them about adverse weather.

It would be possible to use this in other ways though, imagine something like a treasure hunt, where as people arrive at each location, they are simply automatically sent the next clue as a direct message.

Or how about in my case for example I have the primary channel that I broadcast weather updates on less frequently, and a dedicated weather channel that I broadcast updates on much more frequently, BUT no-one knows about the sub channel with the more frequent updates. If however I could announce that the channel exists when a new node arrives on the mesh, then they could decide if they wanted to join that channel or not.

### Describe alternatives you've considered

There are no alternatives.

### Additional context

N/A

broglep, 2025-01-01:
I like this idea / capability to react on new nodes, but there might be some details to be figured out (e.g. if we would want to wait for a neighbour info or not). I think we could look into the mobile apps how they behave and when they notify about a new node being discovered.

I did a little experiment and it is already possible via a small detour to the entity registry and make use of the notification platform (where notify entities are created when they appear)
```

alias: React to new Meshtastic Nodes
description: ""
triggers:
  - trigger: event
    event_type: entity_registry_updated
    event_data:
      action: create
conditions:
  - condition: template
    value_template: >-
      {{ trigger.event.data.entity_id is match('notify\.mesh_node_.*',
      ignorecase=False) }}
actions:
  - action: notify.persistent_notification
    metadata: {}
    data:
      message: >-
        New Node Discovered: {{ state_attr(trigger.event.data.entity_id,
        'friendly_name') }}
      title: Meshtastic
mode: single
```
Maybe you can start with that and try it out to get a better feeling if that already solves some of the usecases

asjmcguire, 2025-01-01:
I will look into this workaround and see how that fits for now.
I had a little think about this over the festive period and I thought, the simplest easiest way to achieve this without having to add a whole new event. If the gateway node had an attribute - probably on the total nodes entity, that was a list of nodes that it knows about. Then I could simply use the api_node_updated event which will definitely fire when a new node appears on the mesh, and I could check whether the nodeid already exists in the known_nodes attribute on total nodes.

broglep, 2025-01-03:
I have the workaround running and seems to work fine (with only exception that it also creates persistent notification for known nodes when you configure the gateway the first time). 

A dedicated event wouldn't be that bad if this is a common use case. And I suspect it could actually be a common use case. As the mobile apps also having a notification when a new node is discovered, this is also a hint that users tend to react to new nodes, so this integration should make it also possible to react to new nodes in an automated way.
  
The approach of having a list of known nodes as part of the gateway is something that will be not good, home assistant warns about this, it will bloat the state history of the entity as the list will be constantly changing

asjmcguire, 2025-01-04:
No no, the state doesn't need to change ONLY the attributes and the attributes only get stored now when the state changes.