meshtastic/home-assistant #22: Way to no create entities in Home Assistant when a new node is found

Ylianst, 2025-01-01:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

This is a wonderful integration, I have been waiting for exactly this sort of integration to remove MQTT. I see over 100 nodes in my area. Is there any way to not create new entities when a new node is found? I am concerned this if flooding my HA with a not of entities with names I can't control (node_xxx). I am also next to a highway and a lot of nodes are visiting and not going to be seen again.

### Describe the solution you'd like

Have a configuration switch to not create nodes when discovered.

### Describe alternatives you've considered

Current MQTT is the alternative, but having a solution that uses the real Meshtastic API is much better.

### Additional context

Thank you so much! I have been looking for exactly this type of integration. An addition to this is to be able to reply to a node that sends a direct/channel message without having the node entity created.

Ylianst, 2025-01-01:
Arg. Found the option. Thanks!

Ylianst, 2025-01-01:
Actually the configuration is to not enable new nodes, but they are still created. I will check to see if they are removed after they timeout of not. If not removed, I could quickly accumulate 100's of node entities.

broglep, 2025-01-01:
I assume you used `System options` -> `Enable newly added entities` right ?

Currently the integration creates notify entities for the full node database (and all new discovered nodes) and the rest of the entities (sensor, device tracker, etc) are only created for the nodes you select as part of the configuration / setup of the integration. 
I did already think about the option to make the notify platform configurable as I'm also not big fan of cluttering home assistant with many entities. I did also think about a service that allows to clear (or expire) the notify entities to address the ever growing nature of the current implementation (which for some might be desired if they want to retain the node database even in case the meshtastic nodes does forget it)

If you know the node id or channel index, you can send messages without the need for the entities, you can use `meshtastic.send_text` service/action in that case. You could use the `meshtastic_api_text_message` event to react on messages (on a channel or private message). But that would be rather an advanced use case

Ylianst, 2025-01-01:
@broglep Thank you so much for your reply. You are exactly correct, I found `System options -> Enable newly added entities` but indeed, after 10 hours, I am already back to 180 entities and this will only go up. Since this option is enabled by default, I have to go manually disable all the entities between integration install and when I get this option disabled.

[image]

I did figure out that I can uninstall/reinstall the integration to clean up the entities, but in my situation, it's going to be a lot of clutter for sure. I do really like being able to create all of the sensors for nodes I select. I have one "mobile" Meshtastic node I want to track, but for all other nodes, I would like no entities at all.

By the way, this is great work. With auto-discovery, clean set of sensors, modern events, this is certainly on the path to what should be the official Meshtastic Home Assistant integration. It feels very professional. I completely disabled MQTT for this use case yesterday and not going back.

I am going to go try `meshtastic.send_text` with a nodeid right now.

broglep, 2025-01-01:
I'm right now working on making the notification platform configurable so that you can select if you would like to create channel and/or node notify targets or not. I think that should also address your use case. There might be even a need for a third option to create notify targets only for selected nodes. That way you could still use the recommended notify platform way of sending messages to the few nodes that you are tracking 

I still struggle a bit with home assistant config flow & options, but once that is sorted out, it should be done quickly.

Ylianst, 2025-01-01:
I have had to work my way up the Home Assistant learning curve myself for my own integration, so, I completely understand. I have used ChatGPT a bit to help out with mixed success.

With Meshtastic adding MQTT flags and control in the latest versions, using MQTT as a bridge to Home Assistant is not a great option anymore. You now have to turn on "allow MQTT" on the sender side to get a message to HA, it's not a good experience and MQTT integration to a bit of a hack. This is a very welcome integration for sure.

I am trying `meshtastic.send_text` and yes, this is exactly perfect. However, when I receive a message, I don't seem to have the senders Meshtastic NodeID. The 'device_id' is the node id of the device that received the message, but I don't have the source node id and so, can't reply. This is a sample message:

```
this:
  entity_id: automation.lora_direct_messages
  state: 'on'
  attributes:
    id: '1735697431879'
    last_triggered: '2025-01-01T17:47:05.651787+00:00'
    mode: parallel
    current: 0
    max: 10
    friendly_name: Lora Direct Messages
  last_changed: '2025-01-01T17:48:46.628786+00:00'
  last_reported: '2025-01-01T17:48:46.628786+00:00'
  last_updated: '2025-01-01T17:48:46.628786+00:00'
  context:
    id: 01JGHFQYN46PDY70J0SYDH2Y2H
    parent_id: null
    user_id: null
trigger:
  id: '0'
  idx: '0'
  alias: null
  platform: device
  event:
    event_type: meshtastic_event
    data:
      device_id: cc4d04df113e51b2d8d86b05a22158ad
      type: message.received
      message: test
      entity_id: meshtastic.gateway_chdu_dm
    origin: LOCAL
    time_fired: '2025-01-01T17:49:06.516500+00:00'
    context:
      id: 01JGHFRJ2MPXM6KB0TJMV03EN7
      parent_id: null
      user_id: null
  description: event 'meshtastic_event'
```

Am I missing anything?

broglep, 2025-01-01:
The `meshtastic_event` is primarily used for device triggers and mostly lives in the home assistant world only (thus only contains device_id or entity_id) and there you have not access to the meshtastic details directly as part of the event (only indirectly via device or entity state that contains the node id)

You need to use the lower level `meshtastic_api_text_message` event. You need to select `Manual event` as trigger in the UI, then you get an event with the following data:
```
"from": packet.from_id,
"to": {"node": to_node, "channel": to_channel},
"gateway": self.get_own_node()["num"],
"message": packet.app_payload,
 ```

Ylianst, 2025-01-01:
Oh wow. Trying it now.

Ylianst, 2025-01-01:
Update to say that I got my automation working perfectly with `meshtastic_api_text_message ` and `meshtastic.send_text`. This enables a lot of capabilities as you can handle text messages and respond with a lot of flexibility. This is great. I personally have a small LLM responding to direct messages from public nodes and have notifications on my phone from HA where there are messages on the primary channel. I can also setup some automation based on the sending node id. So, this is amazing.

broglep, 2025-01-01:
That is great to hear that you got it working. That is maybe something we can add as an (advanced) automation example to https://github.com/broglep/homeassistant-meshtastic?tab=readme-ov-file#examples

Pretty nice use cases you have running in your HA. 

The use case of responding to messages from nodes is something that is covered with device trigger and device actions as long as you know the nodes / devices in advance and select them as devices to be imported to HA (I have such automation in place, where messages from trusted devices trigger things in HA). 
The approach with `meshtastic_api_text_message` and `meshtastic.send_text` now extends this to arbitrary nodes (and without the need to use HA devices / entities at all)

I could resolve the resolve the problem with the config entries options in the meantime. The problem was caused by wrong HA developer documentation, so I'll be able to offer some configuration options for the notification platform in the upcoming release

Ylianst, 2025-01-02:
@broglep Ok, I just created a PR to my attempt at adding advanced instructions for handling text messages from any node. No obligation to accept, but it's working very well for me.

closed 2025-01-02 by broglep

Ylianst, 2025-01-03:
@broglep. Wow, I just tried out 0.4.0 and thank you so much for the update. This dialog box is wonderful.

[image]

I think "Only selected devices" should be the default starting selection if possible. But yes, the dialog and the warning below is outstanding. I went ahead uninstalled, and re-installed with "Only selected devices", all my entities where removed which is great.

When I go into "System options", I would then see this:

[image]

It's confusing in two ways. First, the first option talks about auto-enabling on the first line, but auto-adding on the second. Also, because I opted for only adding selected devices, I expected the first option to be disabled, but it's enabled (??)

In any case, this is an amazing update!!

broglep, 2025-01-03:
I was bit undecided what the default should be, for now I kept the default to match the previous behaviour (to not surprise those who have it already installed) and I have not yet implemented proper config entry migration to ensure that every entry has a valid configuration (some freedom that the alpha stage still allows). And also not yet implemented but planned is a way to prune / cleanup notify entities.  
Because notify platform is recommended way to send messages to any node, a use case I think might be popular one,  I'd rather prefer users using notify mechanisms instead of starting to use the service/action and deal with node id & channel ids without fully understanding the pitfalls. So to facilitate that, all notify entities must be created, hence the current default. 

But "Only selected devices" would be a viable default as well. In that case however the node notify entities do not offer much added value, as you already can send messages to the selected nodes using device action (and don't need notify entities). So if you only ever wanted to send messages to your selected nodes, you could even chose "None of the Nodes".  

I think it needs some more time to get a feeling on how the integration is being used and what users prefer. We can create a poll or discussion at some time in the future to find out more. 

The system options is something that, to my knowledge, a custom component has no influence on, that is home assistant related functionality. The options are indeed confusing, e.g. "Enable newly added entites" talks about entities but then in the description mentioned devices (two separate concepts in home assistant)

I did only quickly read up on that topic, and it is not yet fully clear to me what that options should actually do. What I learned was that one approach with entities is that custom component can add them, but not enable / activate them and then the user is responsible to enable those entities that he wants (instead component adding & enable all of them and the user then disabling those he does not need)

Ylianst, 2025-01-03:
I certainly understand the tradeoffs. I posted my stats below for reference, in the last 60h, I have received 53k packets and relayed 17k with 50 online nodes and 100 total nodes all from public devices. I really like being able to select the nodes I want to create entities for since, I can then identify my own nodes for monitoring, automation, etc. So, there is high value in doing this for nodes you care about and your exactly right, in these cases, the proper use of notify platform is perfect. I also like that I can just select the nodes in home assistant and quickly setup automation (no nodeid). So, it's super high value, but only for the nodes I care about. So, selecting nodes is the best.

I will also point out that you can't really build good automation for a dynamically expanding list of nodes. I guess you could create an automation that when a new node is created, you also dynamically create new automation for it or edit existing automation to add the new node to it, but that seems insane and very much an edge case. I had a similar experience with the RTL_433 integration recently, I started seeing all the devices around my area after a few weeks it was a crazy long list and had to put a stop to it. It's also a unlikely but possible source of spam attack. Regardless, I like you warning message and support any decision you make on this.

"System options is something that, to my knowledge, a custom component has no influence on" - Thanks for that info. Good to know. If I get a chance, I need to report it to the right people.

[image]

commit 7fe9fe35 (2025-01-02): feat: make notification platform configurable

allow to configure if we should create notify entities for channels and nodes

resolves #22
commit 1b6ae43f (2025-01-02): feat: make notification platform configurable

allow to configure if we should create notify entities for channels and nodes

resolves #22