meshtastic/home-assistant #24: Error sending messages

timniklas, 2025-01-02:
### System Health details

## System Information

version | core-2024.12.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4989
Installed Version | 2.0.1
Stage | running
Available Repositories | 1502
Downloaded Repositories | 21

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 109.3 GB
disk_used | 32.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (19.0.0), File editor (5.8.0), Log Viewer (0.17.1), ESPHome Device Builder (2024.12.2), rtl_433 MQTT Auto Discovery (0.8.2), Let's Encrypt (5.2.10), NGINX Home Assistant SSL proxy (3.11.1), Studio Code Server (5.18.0), UniFi Network Application (3.5.0), Mosquitto broker (6.4.1), BosMon (v1.2.9), BosMon RTL-SDR (v0.2.4), Home-Assistant-Matter-Hub (3.0.0-alpha.58), usb-backup (0.0.3), Pocsag RTL-SDR (v0.8.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 4
views | 14
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 22. Dezember 2024 um 20:59
-- | --
current_recorder_run | 2. Januar 2025 um 22:30
estimated_db_size | 504.82 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Sending messages works sometimes, but most of the time i get a connection error.
Sending it via the Meshtastic WebClient works fine tho.

### Reproduction steps

1. Broadcast a message to a channel

### Debug logs

```text
2025-01-02 22:32:18.840 WARNING (MainThread) [custom_components.meshtastic.custom_components.meshtastic.config_flow] Failed to connect to meshtastic device
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 360, in connected_node_ready
    await self._connected_node_ready.wait()
  File "/usr/local/lib/python3.13/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/meshtastic/api.py", line 135, in connect
    await asyncio.wait_for(self._interface.connected_node_ready(), timeout=30)
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/meshtastic/config_flow.py", line 125, in validate_input_for_connection
    async with MeshtasticApiClient(
               ~~~~~~~~~~~~~~~~~~~^
        data,
        ^^^^^
    ...<2 lines>...
        no_nodes=no_nodes,
        ^^^^^^^^^^^^^^^^^^
    ) as client:
    ^
  File "/config/custom_components/meshtastic/api.py", line 222, in __aenter__
    await self.connect()
  File "/config/custom_components/meshtastic/api.py", line 149, in connect
    raise MeshtasticApiClientCommunicationError from e
custom_components.meshtastic.api.MeshtasticApiClientCommunicationError
2025-01-02 22:37:59.605 ERROR (MainThread) [homeassistant.components.automation.meshtastic_chatgpt] Meshtastic ChatGPT: Error executing script. Error for call_service at pos 2: Connection interrupted
2025-01-02 22:37:59.611 ERROR (MainThread) [homeassistant.components.automation.meshtastic_chatgpt] Error while executing automation automation.meshtastic_chatgpt: Connection interrupted
```

asjmcguire, 2025-01-02:
Can confirm that I had the same problem - but https://github.com/broglep/homeassistant-meshtastic/issues/12#issuecomment-2565483930 this has so far fixed all the problems.

broglep, 2025-01-02:
@timniklas Please try with the new released version 0.4.0 that has some improvements in this area. 

As you are mentioning the web client, please make sure that the web client is closed when using this integration as meshtastic firmware only allows a single connection. If you have both web client as well as the integration running, they will fight for the connection and cause recurring connection interupts

timniklas, 2025-01-02:
@broglep Unfortunately, the replies are still not sending reliably.
But at least the requests seem to arrive in the integration much more reliable.

timniklas, 2025-01-02:
> Can confirm that I had the same problem - but [#12 (comment)](https://github.com/broglep/homeassistant-meshtastic/issues/12#issuecomment-2565483930) this has so far fixed all the problems.

My solution for now is to switch to a serial connection, instead of wifi

broglep, 2025-01-03:
> My solution for now is to switch to a serial connection, instead of wifi
Will be interested to hear your experience with serial. In my test setups it is less stable than TCP.
In terms of reliably sending responses to message, not only the connection between home assistant and meshtastic node itself can cause problems, in my experience the firmware itself sometimes causes a response not be be sent (e.g. if radio is too busy). Make sure that you have a delay in home assistant before sending a reply. Serial in that case could actuality have a positive side effect because serial interface is slower in my tests and thus increases likelihood that the message is sent after the radio busy period.   

If you switch to TCP and start experiencing issues again with tcp, can you try to capture the serial log output so that we can gain some more insight on why the connection might be interrupted?

timniklas, 2025-01-04:
I've switched to a new simple node with BLE switched of and connected via a serial connection.
Everything works fine now, I dont event need a delay for the responses.
TCP connections seems to be the issue.

[image]
[image]

closed 2025-03-15 by timniklas