meshtastic/home-assistant #25: Unable to ignore devices

timniklas, 2025-01-03:
### System Health details

## System Information

version | core-2024.12.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.0
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4971
Installed Version | 2.0.1
Stage | running
Available Repositories | 1502
Downloaded Repositories | 21

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 109.3 GB
disk_used | 32.7 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (19.0.0), File editor (5.8.0), Log Viewer (0.17.1), ESPHome Device Builder (2024.12.2), rtl_433 MQTT Auto Discovery (0.8.2), Let's Encrypt (5.2.10), NGINX Home Assistant SSL proxy (3.11.1), Studio Code Server (5.18.0), UniFi Network Application (3.5.0), Mosquitto broker (6.4.1), BosMon (v1.2.9), BosMon RTL-SDR (v0.2.4), Home-Assistant-Matter-Hub (3.0.0-alpha.58), usb-backup (0.0.3), Pocsag RTL-SDR (v0.8.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 4
views | 14
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 22. Dezember 2024 um 20:59
-- | --
current_recorder_run | 3. Januar 2025 um 00:34
estimated_db_size | 504.82 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

My bluetooth node keeps showing up in home assistant (i dont want to use it there).
Unfortunately I am unable to ignore ist like with other integrations.

### Reproduction steps

[image]

### Debug logs

```text
none
```

broglep, 2025-01-03:
There used to be an ignore option, but I also can't see it anymore with recent home assistant version, will have to investigate why this stopped working

asjmcguire, 2025-01-03:
made all the more irritating by the fact that I keep getting a notification that new devices were discovered on the network, multiple times a day...

broglep, 2025-01-03:
I think I have a acceptable solution for this. This has been broken as part of change on how configuration flow unique id are set. If the integration does not set a unique id, you are not able to ignore it. This integration uses the node id as the unique id, but that can be only retrieved when making a connection. But as a new second, connection breaks previous ones (or is not possible at all with bluetooth or serial), auto discovery can't automatically open a connection. 

But based on my tests it is possible to change unique id of the flow in the middle, so I can first set a "weak" unique id like bluetooth address or ip address so that you can ignore that, and then at the second stage where you start to configure the discovered config entry, set the actual node id. Downside is that only at that stage the integration can detect that you already have the node configured (e.g. if you have gateway via TCP configured and then via auto discovery add another entry for the serial interface of that gateway) 

For bluetooth this should solve the issue as the mac should be stable, but for tcp it could still produce recurring config entries that all need to be ignored when meshtastic gateway receives different ip addresses.  

The best solution would be if meshtastic firmware would allow multiple connections. That way we could always get the node id and new auto discovered entries would never appear at all when a gateway is already configured

timniklas, 2025-01-04:
An fix for that issue would be great, it keeps getting worse :D

[image]

broglep, 2025-01-04:
@timniklas you can try out the fix by replacing `config_flow.py` with dev version from https://github.com/broglep/homeassistant-meshtastic/commit/0897d1fcb19973a601e35eb13ba0cf7253f8f24b

timniklas, 2025-01-04:
> @timniklas you can try out the fix by replacing `config_flow.py` with dev version from [0897d1f](https://github.com/broglep/homeassistant-meshtastic/commit/0897d1fcb19973a601e35eb13ba0cf7253f8f24b)

I tried it, but unfortunately it didn't change anything

timniklas, 2025-01-04:
> > @timniklas you can try out the fix by replacing `config_flow.py` with dev version from [0897d1f](https://github.com/broglep/homeassistant-meshtastic/commit/0897d1fcb19973a601e35eb13ba0cf7253f8f24b)
> 
> I tried it, but unfortunately it didn't change anything

Nvm restarted HA a few times and now I am able to ignore these entities 👍

commit 0897d1fc (2025-01-04): fix: allow to ignore auto discovered entries

ensure that we always set unique id

fixes #25
closed 2025-01-18 by broglep