meshtastic/home-assistant #26: Getting the Meshtastic client connected while this integration is running

Ylianst, 2025-01-04:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

One thing I realized using this integration that I did not expect is that Meshtastic only allows 1 client to connect at any given time. So, connecting the Meshtastic web page will apparently disconnect this integration so, only one can connect at any given time. This was not the behavior with MQTT. However, I absolutely love this integration, so I am sticking with it but I now need a UI for manual chatting with others on the Meshtastic network.

Before asking for anything, how is everyone dealing with this? Do you use a second Meshtastic device?

### Describe the solution you'd like

One crazy idea would be to have this integration offer up a public TCP port for the Meshstatic client to connect to. I realize this would be somewhat complex to do, but could end up having the client be an extra tab on the side bar. Creating a Meshtastic proxy port would be some work, so, not asking for this but the proxy would possibly be a lot faster then the LORA device and (depending on the implementation) could be secured/authenticated and accessed anywhere Home Assistant is accessible.

### Describe alternatives you've considered

Right now, I think I may work on a little chat UI within HA. I could use a second device but that counts as a "hop" so, not idea. Still figuring it out.

### Additional context

Feedback from anyone encouraged.

Ylianst, 2025-01-04:
Ok, I created myself two text helpers. One with the sender's nodeid and last message on the primary channel and another one to send a message on the primary channel. It's very simple but will work for now. I get notified on my phone when there is traffic on the primary channel and this way I can respond.

[image]

I needed 2 helpers and 2 automation for this to work.

broglep, 2025-01-04:
I use dedicated meshtastic nodes as gateways and connect only them to home assistant (prefer Raspberry Pi Zeros with HAT for this use case). For personal use & messaging I use separate nodes (as CLIENT_MUTE) and for administration the mesh also dedicated admin node (CLIENT_HIDDEN), and when I have a router, it is also separate node. I use a gateway (as CLIENT) next to a router as a possibility for internet based backhaul as well as collecting mesh metrics visible from that particular site. I don't like to mix use cases and with current stability of the firmware, it might be even risky (like putting all eggs into one basket). Seen quite some strange things where nodes got stuck or lost part of its data / configuration. 

The additional hops are indeed a bit of an issue, but in my case I could compensate for it by increasing the max hop (but didn't yet have to). As I'm in a low density area and quite in control of the mesh topology, increasing the max hop will almost have no negative side effect.  

Building a simple user interface (e.g. custom card or even own section like e.g. todo list does) should be doable in home assistant, but I'm not that experienced with the UI side of things in home assistant. But it should remain simple. I already thought about custom user interface as part of the message recording feature, but in the end decided to rather re-use existing building blocks from home assistant and thus have used logbook as the interface to display channel message history (you can filter for the channel entity, then you are pretty close to a chat history UI)
For more advanced needs, another possibility might be to embed the browser interface into home assistant.

I kind of like the idea to enable regular other meshtastic clients to connect to home assistant and the integration then proxing or even "multiplexing" to the single connection to the device. Maybe I can experiment with this a bit, but it could very well uncover some roadblocks inside the firmware and/or the clients. 

Unfortunately nothing is possible right away with the current capabilities of the code/integration, as browser interface does not use TCP itself but a Websocket variation the protocol and exposing a bluetooth peripheral bluetooth might be tricky due to hardware dongle involvement. Python cli using tcp is probably not that nice of a user interface as well. Websocket is probably most promising for a PoC, and once a foundation is present, should be possible to extend it to other interfaces

Maybe slightly related to this: I'm still interested into trying out if mqtt could be offered as a forth way to connect to a node. But I'm unsure if will fit nicely into the existing code as mqtt interface is pretty different to the others and as such unsure if it should be part of the scope of this integration. But it would enable some nice use cases, like collecting metrics for certain nodes on a shared/global mqtt broker.

asjmcguire, 2025-01-04:
I would definitely suggest that we should probably look at the possibility of a dedicated sqlite database (probably in the share directory) for storing message data in, rather than using the logbook - just because obviously in heavy traffic areas the use of logbook for channel messages etc will significantly increase the size of the Home Assistant database.

broglep, 2025-01-04:
> I would definitely suggest that we should probably look at the possibility of a dedicated sqlite database (probably in the share directory) for storing message data in, rather than using the logbook

Logbook is exactly optimised for that and uses recorder integration to be able to use something else then the default sqlite database. It also makes it possible to exclude channel messages altogether, or only exclude some high volume channels. Can't foresee a good reason to introduce our own storage solution, I would alway prefer using my recorder database instead of a custom components own managed sqlite database for multiple reasons. 
If I'm dealing with lot of data, chances are high that already have capable recorder database in place and that is far better than using a custom components own sqlite database. There are also many other benefits in terms of maintainability when using a single database instead of each component having its own homegrown storage. 

The only reason I see to have own storage solution would be when we'd want to store message in a more structured way to enable some advanced filtering/querying capabilities, like if you want to create a longterm meshtastic channel message archive, but that is outside the scope of this integration.

Ylianst, 2025-01-05:
Thanks @broglep  and @asjmcguire for your input. I am staying away from MQTT as it now has limitations flags associated with it in the latest firmware. Yes, going forward there are a lot of options: 

- Meshtastic chat card.
- A HTTP proxy (multiplex) for the Meshtastic web client.
- Using another node with an extra hop (CLIENT_MUTE).
- A new web client page within Home Assistant.
- Using MQTT (New firmware have limitation flags).
- Using existing helpers and automation to create a tiny client.

If anyone else as other options, please post. Regardless, there are a lot of exciting options going forward.

broglep, 2025-01-05:
I have completed a PoC that allows to open meshtastic web frontend and then connect it to home assistant. After that the integration does forward the packets to the web frontend. Is even seems to work ok when you have two web frontends open, they can both be served by home assistant. The only thing is that for messages sent by the gateway, either initiated by one of the web clients or home assistant itself won't appear in the web frontend.

I haven't tested it yet, but this should also work when mixing connection types. So you can connect via web frontend to nodes that are connected via bluetooth or serial

[image]

Ylianst, 2025-01-14:
OMG!!! This is going to be amazing!! This is the uncontested king of Meshtastic integrations! I just updated to 0.4.0, but I want to give this a try so bad.

closed 2025-01-18 by broglep

Ylianst, 2025-01-18:
This is HUGE! So far, just going from 0.4.0 to 0.5.0 did not work for me. It's saying the configuration is wrong. I am going to try to fully remove and re-install again.

asjmcguire, 2025-01-18:
> This is HUGE! So far, just going from 0.4.0 to 0.5.0 did not work for me. It's saying the configuration is wrong. I am going to try to fully remove and re-install again.

Same issue as here: https://github.com/broglep/homeassistant-meshtastic/issues/34

broglep, 2025-01-18:
FYI: The feature is disabled by default (due to experimental nature and security implications), so you need to go to the meshtastic integration config and enable it for existing gateways

commit e7df5405 (2025-01-18): feat: support meshtastic web client

fixes #26
commit ab0f14df (2025-01-18): feat: support meshtastic web client

fixes #26