meshtastic/home-assistant #34: Alpha v0.5.0 cannot be setup

asjmcguire, 2025-01-18:
### System Health details

## System Information

version | core-2025.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4976
Installed Version | 2.0.3
Stage | running
Available Repositories | 1506
Downloaded Repositories | 40

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 20 November 2025 at 00:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | eu-central-1-11.ui.nabu.casa
certificate_status | ready
instance_id | d74dbc9c9b6b4563b25a3c274331fe7a
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 14.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (7.0.0), Home Assistant Google Drive Backup (0.112.1), File editor (5.8.0), Samba share (12.4.0), ESPHome Device Builder (2024.12.2), Terminal & SSH (9.16.0), Whisper (2.4.0), Piper (1.5.2), Auto Guest Login (1.4.0), Node-RED (19.0.0), Music Assistant Server (beta) (2.4.0b16), Govee to MQTT Bridge (2025.01.04-2c39a50f), OpenThread Border Router (2.13.0), chrony (5.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 11
-- | --
resources | 18
views | 64
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 7 January 2025 at 15:01
-- | --
current_recorder_run | 18 January 2025 at 15:09
estimated_db_size | 3285.66 MiB
database_engine | mysql
database_version | 10.7.8

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Home Assistant says the integration cannot be setup

### Reproduction steps

Update to v0.5.0 from Home Assistant updates system

### Debug logs

```text
Detected blocking call to import_module with args ('custom_components.meshtastic',) in /usr/src/homeassistant/homeassistant/loader.py, line 1075: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)
Detected blocking call to import_module with args ('custom_components.meshtastic.config_flow',) in /usr/src/homeassistant/homeassistant/loader.py, line 1304: return importlib.import_module(f"{self.pkg_path}.{platform_name}") inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/util/async_.py", line 112, in sem_task return await task File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1315, in async_init flow, result = await self._async_init(flow_id, handler, context, data) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1340, in _async_init flow = await self.async_create_flow(handler, context=context, data=data) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 1536, in async_create_flow handler = await _async_get_flow_handler( File "/usr/src/homeassistant/homeassistant/config_entries.py", line 3333, in _async_get_flow_handler await _load_integration(hass, domain, hass_config) File "/usr/src/homeassistant/homeassistant/config_entries.py", line 3312, in _load_integration await integration.async_get_platform("config_flow") File "/usr/src/homeassistant/homeassistant/loader.py", line 1108, in async_get_platform platforms = await self.async_get_platforms((platform_name,)) File "/usr/src/homeassistant/homeassistant/loader.py", line 1173, in async_get_platforms platforms.update(self._load_platforms(platform_names)) File "/usr/src/homeassistant/homeassistant/loader.py", line 1098, in _load_platforms platform_name: self._load_platform(platform_name) File "/usr/src/homeassistant/homeassistant/loader.py", line 1272, in _load_platform cache[full_name] = self._import_platform(platform_name) File "/usr/src/homeassistant/homeassistant/loader.py", line 1304, in _import_platform return importlib.import_module(f"{self.pkg_path}.{platform_name}")

Error occurred loading flow for integration meshtastic: cannot import name 'defaultdict' from 'pydantic.utils' (/usr/local/lib/python3.13/site-packages/pydantic/utils.py)
```

broglep, 2025-01-18:
That is strange, can you try run `python3 -c 'from collections import defaultdict; print(defaultdict(list))'
` in your home assistant python environment? 

The output of `pip3 freeze` might also help to find out why this is happening

asjmcguire, 2025-01-18:
```
defaultdict(<class 'list'>, {})
```

```
accuweather==4.0.0
acme==3.0.1
adax==0.4.0
Adax-local==0.1.5
adb-shell==0.4.4
adext==0.4.3
adguardhome==0.7.0
advantage-air==0.4.4
AEMET-OpenData==0.6.4
aenum==3.1.15
afsapi==0.2.7
agent-py==0.0.24
aio-geojson-geonetnz-quakes==0.16
aio-geojson-geonetnz-volcano==0.9
aio-geojson-nsw-rfs-incidents==0.7
aio-geojson-usgs-earthquakes==0.3
aio-georss-client==0.13
aio-georss-gdacs==0.10
aio_geojson_client==0.21
aio_geojson_generic_client==0.5
aioacaia==0.1.13
aioairq==0.4.3
aioairzone==0.9.7
aioairzone-cloud==0.6.10
aioambient==2024.8.0
aioapcaccess==0.4.2
aioaquacell==0.2.0
aioaseko==1.0.0
aioasuswrt==1.4.0
aioautomower==2025.1.0
aioazuredevops==2.2.1
aiobafi6==0.9.0
aiobotocore==2.13.1
aiocache==0.12.3
aiocoap==0.4.12
aiocomelit==0.10.1
aioconsole==0.8.1
aiodhcpwatcher==1.0.2
aiodiscover==2.1.0
aiodns==3.2.0
aiodukeenergy==0.2.2
aioeafm==0.1.2
aioeagle==1.1.0
aioecowitt==2024.2.1
aioelectricitymaps==0.4.0
aioemonitor==1.0.5
aioesphomeapi==28.0.0
aiofile==3.9.0
aiofiles==24.1.0
aioflo==2021.11.0
aioftp==0.21.3
aiogithubapi==24.6.0
aioguardian==2022.7.0
aiohappyeyeballs==2.4.4
aioharmony==0.2.10
aiohasupervisor==0.2.2b5
aiohomekit==3.2.7
aiohttp==3.11.11
aiohttp-cors==0.7.0
aiohttp-fast-zlib==0.2.0
aiohttp-retry==2.9.1
aiohttp-sse-client2==0.3.0
aiohttp_socks==0.10.1
aiohue==4.7.3
aioimaplib==1.1.0
[... 1375 lines cut ...]
yeelight==0.7.14
yeelightsunflower==0.0.10
yolink-api==0.4.7
youless_api==2.1.2
youtubeaio==1.1.5
yt-dlp==2024.12.23
zabbix-utils==2.0.2
zamg==0.3.6
zeep==4.3.1
zengge==0.2
zeroconf==0.136.2
zeversolar==0.3.2
zha==0.0.45
zha-quirks==0.0.130
zhong-hong-hvac==1.0.13
ziggo-mediabox-xl==1.1.0
zigpy==0.74.0
zigpy-deconz==0.24.1
zigpy-xbee==0.21.0
zigpy-zigate==0.13.2
zigpy-znp==0.13.1
zm-py==0.5.4
zwave-js-server-python==0.60.0
zwave-me-ws==0.4.3
```

Ylianst, 2025-01-18:
I am also getting an error installing 0.5.0. Even a full remove/re-install.

[image]

broglep, 2025-01-18:
This is caused by one instance of wrong `defaultdict` import, that happened to work on older home assistant versions, but not newer ones. A fixed version is about to be released

closed 2025-01-18 by broglep

asjmcguire, 2025-01-18:
Confirmed working. Web interface works. Love it. Cheers.

Ylianst, 2025-01-18:
Arg. 0.5.1 is telling me the device is already configured, but I don't see the integration active. Anyway to flush previous data? (Using SSH is not a problem).

[image]

commit 547524c0 (2025-01-18): fix: defaultdict import

fixes #34