meshtastic/home-assistant #35: Last Heard time is a bit.... odd....

asjmcguire, 2025-01-18:
### System Health details

## System Information

version | core-2025.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4976
Installed Version | 2.0.3
Stage | running
Available Repositories | 1506
Downloaded Repositories | 40

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 20 November 2025 at 00:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | eu-central-1-11.ui.nabu.casa
certificate_status | ready
instance_id | d74dbc9c9b6b4563b25a3c274331fe7a
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 14.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (7.0.0), Home Assistant Google Drive Backup (0.112.1), File editor (5.8.0), Samba share (12.4.0), ESPHome Device Builder (2024.12.2), Terminal & SSH (9.16.0), Whisper (2.4.0), Piper (1.5.2), Auto Guest Login (1.4.0), Node-RED (19.0.0), Music Assistant Server (beta) (2.4.0b16), Govee to MQTT Bridge (2025.01.04-2c39a50f), OpenThread Border Router (2.13.0), chrony (5.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 11
-- | --
resources | 18
views | 64
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 7 January 2025 at 15:01
-- | --
current_recorder_run | 18 January 2025 at 15:09
estimated_db_size | 3285.66 MiB
database_engine | mysql
database_version | 10.7.8

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

My last heard time, is 12 hours ago for remote nodes, but looking at the logbook.....

[image]

We can see that actually it was heard 12 hours AFTER the time the integration claims.

Eg:

Actually heard at 20:29:38 but the integration claims 08:29

The web interface also claims 12 hours ago too. The Meshtastic app says 5 minutes ago.

### Reproduction steps

Observe the last heard time. Looks like it's just an issue with 24 hour time.

### Debug logs

```text
No logs unless if you really need them.
```

asjmcguire, 2025-01-18:
[image]

OK so looking back through the logbook, it's not just a simple ampm issue, it's consistently 12 hours BEHIND as you can see from the logbook, when it was actually AM in GMT/UTC which should render a simple 24 hour clock issue temporarily fixed, it is in fact as you can see still claiming the last heard time was 12 hours before the recorded time.

At 11:56:11am on the 18th January, the integration claims last heard was 23:56 on 17th January

broglep, 2025-07-06:
Feel free to reopen if still an issue

closed 2025-07-06 by broglep