meshtastic/home-assistant #37: Bluetooth PIN Support

fti7, 2025-01-20:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Please consider to add PIN Support for Bluetooth Connections (esphome Proxy Mode)

Connect Device via Bluetooth.
IMPORTANT: PIN is not supported, please remove it first.

### Describe the solution you'd like

PIN Form when adding Device

### Describe alternatives you've considered

-

### Additional context

-

broglep, 2025-01-20:
Out of the box support for bluetooth PIN is currently not really feasible due to limitation of the third party libraries & projects. For reference see:
- https://github.com/esphome/feature-requests/issues/2013
- https://github.com/hbldh/bleak/issues/827

For proper bluetooth PIN support in this integration, bleak will have to provide an PIN API, and ESP Home bluetooth proxy & its home assistant integration would have to implement it.
 
But you are not fully out of luck, there are two possible solutions you could explore: 

1. You can try ESPHome with a custom config and hardcode the PIN into the esphome proxy device. Check https://esphome.io/components/ble_client.html#on-passkey-request on how to do this. With this approach the esphome proxy device will provide the PIN and it will be fully transparent to this integration (work same way as device would not have a PIN). I have not yet tested this and I've so far not found any confirmation online that someone has got this working, only found people asking for help to get it working 

2. If you don't have to use ESPHome bluetooth proxy, you could switch to bluetooth dongle and I could build a experimental version of this integration with bluetooth PIN support based on https://github.com/hbldh/bleak/pull/1100. I already did some experiments with this and it was kind of working, but only if you don't have any other [pairing agents](https://technotes.kynetics.com/2018/pairing_agents_bluez/) on your linux system. If this will be a viable solution depends on your system you are running home assistant in

In general in my tests, the whole bluetooth pairing was quite difficult to working reliably, apart from the rather experimental support from the projects, I also suspect some issues on meshtastic firmware side. 
For approach 2 I had occasionally to fully remove all pairings in bluez and then re-pair to get the connection working again.

williamfligor, 2025-06-26:
I was able to get a ESPHome Bluetooth Proxy connected to my T1000-e with a bluetooth PIN using [this ESPHome config](https://gist.github.com/williamfligor/28eb01269c708f96377779e15afa96cd).

What got it working was on connection setting the encryption:

```yaml
ble_client:
  - mac_address: <MAC ADDRESS>
    id: ble_itag

    # Let Home Assistant initiate the connection
    auto_connect: false

    # When connected force the BLE connection to be encrypted
    on_connect:
      then:
        - lambda: |-
            esphome::ble_client::BLEClient* client = id(ble_itag);
            esp_ble_set_encryption(client->get_remote_bda(), ESP_BLE_SEC_ENCRYPT_MITM);

    # After encryption is set, the device will ask for the pin
    on_passkey_request:
      then:
        - ble_client.passkey_reply:
            id: ble_itag
            passkey: <PIN NUMBER>
```

awkaplan, 2025-08-22:
> I was able to get a ESPHome Bluetooth Proxy connected to my T1000-e with a bluetooth PIN using [this ESPHome config](https://gist.github.com/williamfligor/28eb01269c708f96377779e15afa96cd).
> 
> What got it working was on connection setting the encryption:
> 
> ble_client:
>   - mac_address: <MAC ADDRESS>
>     id: ble_itag
> 
>     # Let Home Assistant initiate the connection
>     auto_connect: false
> 
>     # When connected force the BLE connection to be encrypted
>     on_connect:
>       then:
>         - lambda: |-
>             esphome::ble_client::BLEClient* client = id(ble_itag);
>             esp_ble_set_encryption(client->get_remote_bda(), ESP_BLE_SEC_ENCRYPT_MITM);
> 
>     # After encryption is set, the device will ask for the pin
>     on_passkey_request:
>       then:
>         - ble_client.passkey_reply:
>             id: ble_itag
>             passkey: <PIN NUMBER>

Thank you @williamfligor for this workaround.  After some tweaking, I discovered one minor change depending on your device and bluetooth configuration.  For people using an nRF52 device with a fixed pin, you must use `ESP_BLE_SEC_ENCRYPT_NO_MITM`:

```
on_connect:
  then:
    - lambda: |-
            esphome::ble_client::BLEClient* client = id(ble_itag);
            esp_ble_set_encryption(client->get_remote_bda(), ESP_BLE_SEC_ENCRYPT_NO_MITM);
```