meshtastic/home-assistant #38: Reboots every 346 / 347 seconds using wifi

madmic1314, 2025-01-22:
### System Health details

## System Information

version | core-2025.1.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.3
Stage | running
Available Repositories | 1506
Downloaded Repositories | 23

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 48

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 28 October 2025 at 00:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | eu-central-1-8.ui.nabu.casa
certificate_status | ready
instance_id | 4c10c4e229f64a44b168e793ed1f5bc3
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 109.3 GB
disk_used | 55.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | AdGuard Home (5.2.2), MariaDB (2.7.2), Mosquitto broker (6.5.0), Node-RED (19.0.0), Samba share (12.4.0), Terminal & SSH (9.16.0), texecom2mqtt (1.2.3), ESPHome Device Builder (2024.12.4), Zigbee2MQTT (2.0.0-2), File editor (5.8.0), Rclone Backup (3.3.4), Let's Encrypt (5.2.10), Tailscale (0.24.0), Uptime Kuma (0.12.4), Advanced SSH & Web Terminal (20.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 10
views | 32
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 10 January 2025 at 18:15
-- | --
current_recorder_run | 22 January 2025 at 11:41
estimated_db_size | 3189.42 MiB
database_engine | mysql
database_version | 10.11.6

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Enabled this integration - Healtec v3 started rebooting every 346 / 347 seconds. Tried removing powr and reset, but the same.

[image]

### Reproduction steps

1. Install integration
2. Connect to Meshtastic via wifi (same SSID & subnet as HA)
3. Wait 346 / 347 seconds
...

[home-assistant_meshtastic_2025-01-22T14-07-15.235Z.log](https://github.com/user-attachments/files/18506656/home-assistant_meshtastic_2025-01-22T14-07-15.235Z.log)

[home-assistant_2025-01-22T14-08-27.462Z.log](https://github.com/user-attachments/files/18506659/home-assistant_2025-01-22T14-08-27.462Z.log)

### Debug logs

```text
Attached as "home-assistant_2025-01-22T14-08-27.462Z.log" in reproduction steps as larger than the limit here.
```

### Diagnostics dump

Attached as "home-assistant_meshtastic_2025-01-22T14-07-15.235Z.log in reproduction steps as larger than the limit here.

broglep, 2025-01-22:
What meshtastic firmware version are you using? I don't think this integration is solely responsible for the reboots. 

I know that recently some memory leaks were fixed in the meshtastic firmware that caused crashes after a certain time. There are also numerous other stability issues with the firmware I experienced (e.g. long running http requests can cause the watchdog to trigger and cause a reboot the device)

To further investigate, we would need device serial logs

madmic1314, 2025-01-22:
Was on v2.5.3 - upgraded this to 2.5.18 and it seems ok now. Will monitor and report back there are any other issue

closed 2025-01-22 by madmic1314

broglep, 2025-01-22:
Ok, great to hear. v2.5.3 is rather old and there were some improvements since then (I don't have full overview of firmware changes, lost a bit track of them, but the memory leak fix was pretty sure after v2.5.3)

If you have the possibility, make sure to capture the firmware device logs, they are quite helpful

infinitytec, 2025-03-01:
Had this issue too. Thanks for the tip!

[image]