meshtastic/home-assistant #41: Unable to add meshtastic node when using unifi integration

lillpjot, 2025-02-05:
### System Health details

## System Information

version | core-2025.1.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.73-haos
arch | x86_64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4982
Installed Version | 2.0.5
Stage | running
Available Repositories | 1524
Downloaded Repositories | 63

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Swedish Public Transport Sensor (HASL)</summary>

Version | 3.1.1
-- | --
Schema | 3
Instances | 3
Database Size | 344 bytes
Startup in progress | false
Running tasks | false

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.02.0
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 62.3 GB
disk_used | 24.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Studio Code Server (5.18.1), Mosquitto broker (6.5.0), Node-RED (19.0.0), Terminal & SSH (9.16.0), Z-Wave JS UI (3.19.0), MariaDB (2.7.2), Advanced SSH & Web Terminal (20.0.0), ESPHome Device Builder (2024.12.4), phpMyAdmin (0.11.0), Home Assistant Google Drive Backup (0.112.1), Zigbee2MQTT (2.0.0-2), Spotify Connect (0.13.0), Matter Server (7.0.0), PSA Car Controller (v3.5.3), Piper (1.5.2), Whisper (2.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 29
views | 33
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25 January 2025 at 09:05
-- | --
current_recorder_run | 5 February 2025 at 22:38
estimated_db_size | 5774.50 MiB
database_engine | mysql
database_version | 10.11.6

</details>

<details><summary>Sonoff</summary>

version | 3.8.2 (c4b6fda)
-- | --
cloud_online | 0 / 1
local_online | 0 / 0

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Svensk Postutdelning</summary>

Version | 1.0.7
-- | --
Idle | false
Instances | 1

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

When trying to connect to a Meshtastic (Heltec V3) device it fails in some part of the intializing.
It is part of the config, but says "Failed to setup: check logs" and is the marked red.
I can see statistics in the GUI, and it is updated.
But I'm unable to receive messages or send them on the public channel.
Each time I try reload it get the same message.

After looking in the logs I found a message that the device is already registred and it is registred by Unifi integration , based on the MAC address. 

Therefore it seems like it doesnt handle that Unifi is before and registrering the device.

I tried to delete the Unifi device in the core.entity_registry but Unifi seems to be quick to recreate it.

Unfortunately I do not have a installation without Unifi. 

I have tried with uninstalling the integration, restart and the install all restart again. Same error.

### Reproduction steps

1. Install Unifi integration 
2. Install the integration with HACS
3. Restart HASS
4. Enable / connect a Meshtastic 
5. The Failed Error message arrives and message to/from Lora fails....

### Debug logs

```text
Logger: homeassistant.config_entries
Source: config_entries.py:640
First occurred: 22:39:43 (4 occurrences)
Last logged: 23:03:37

Error setting up entry Westerhouds Nälsta for meshtastic
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/meshtastic/__init__.py", line 133, in async_setup_entry
    await _setup_meshtastic_devices(hass, entry, client)
  File "/config/custom_components/meshtastic/__init__.py", line 161, in _setup_meshtastic_devices
    await _setup_meshtastic_device(
        client, device_hardware_names, device_registry, entry, gateway_node, node, node_id
    )
  File "/config/custom_components/meshtastic/__init__.py", line 253, in _setup_meshtastic_device
    device_registry.async_update_device(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        d.id,
        ^^^^^
        new_connections=connections,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        via_device_id=None if via_device is None else UNDEFINED,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 992, in async_update_device
    new_values["connections"] = self._validate_connections(
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~^
        device_id, new_connections, False
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 1080, in _validate_connections
    raise DeviceConnectionCollisionError(
        normalized_connections, existing_device
    )
homeassistant.helpers.device_registry.DeviceConnectionCollisionError: Connections {('mac', '34:b7:da:63:xx:xx')} already registered with DeviceEntry(area_id=None, config_entries={'f2cf614b9d875f199e6a648a668xxxx'}, configuration_url=None, connections={('mac', '34:b7:da:63:xx:xx')}, created_at=datetime.datetime(2025, 2, 4, 21, 59, 50, 352940, tzinfo=datetime.timezone.utc), disabled_by=None, entry_type=None, hw_version=None, id='374856f059fde03dca795bca83e3xxxx', identifiers=set(), labels=set(), manufacturer='Espressif Inc.', model=None, model_id=None, modified_at=datetime.datetime(2025, 2, 4, 21, 59, 50, 352958, tzinfo=datetime.timezone.utc), name_by_user=None, name='esp32s3-63xxxx', primary_config_entry=None, serial_number=None, suggested_area=None, sw_version=None, via_device_id=None, is_new=False, _cache={'json_repr': b'{"area_id":null,"configuration_url":null,"config_entries":["f2cf614b9d875f199e6a648a6683xxxx"],"connections":[["mac","34:b7:da:63:xx:xx"]],"created_at":1738706390.35294,"disabled_by":null,"entry_type":null,"hw_version":null,"id":"374856f059fde03dca795bca83e3xxxx","identifiers":[],"labels":[],"manufacturer":"Espressif Inc.","model":null,"model_id":null,"modified_at":1738706390.352958,"name_by_user":null,"name":"esp32s3-63xxxx","primary_config_entry":null,"serial_number":null,"sw_version":null,"via_device_id":null}'})
```

broglep, 2025-02-05:
Can you share the full logs after a start of home assistant?

Not really sure if this integration is responsible for that issue or unify. If I understand it correctly, somehow the unify integration did already "claim" the mac address. I'm not familiar with that integration and can't test as I don't own unify devices. Do you have an idea why the meshtastic node mac address might be already used by unify? Does the unify integration register all known ethernet devices in home assistant with their mac addresses?

How do you setup the meshtastic node, is it via tcp/ip?

lillpjot, 2025-02-05:
I use TCP/IP,  Unifi integration is a integration to the Unifi network controller and all my systems that gets an IP is automatically registered in HASS (linked to integration below), so all WIFI and ethernet devices are added.
I believe that the Unifi integration will always be before since the Meshtastic device will need to get IP before it can communicate on TCP/IP. 

I do not know the correct way to share device information or if it should be shared between integrations. 
I logged the error here mostly because I've used Unifi for several years and this was newer. I will to try to reach out to the developer of Unifi integration and see if he can help bring som clarifications also.

https://www.home-assistant.io/integrations/unifi/

I will not be able to get the full debug until friday morning (CET).

lillpjot, 2025-02-07:
Ok, did a restart and picked up the log,
Still the most prominent error I could find is the error on the DeviceConnectionCollisionError (see below and attached file).

Had a quick exchange with the Unifi developer and the Unifi integration is part of core and there are several other integrations that shares devices with Unifi for a while without known issues. 

Also, at the moment it is just my speculation that there is a conflict/miss communication in the device registry that stops the complete setup of meshtastic and hinders message handling, I can see the device and get stats, but are not able to send/recive messages and the integration says failed to setup. 

Unfortunately I'm not able to uninstall the Unifi integrations since I use it for a couple of automations for controlling the family teens access.

Thanks for looking at this.

```

2025-02-07 17:16:53.699 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Westerhouds Nälsta for meshtastic
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/meshtastic/__init__.py", line 133, in async_setup_entry
    await _setup_meshtastic_devices(hass, entry, client)
  File "/config/custom_components/meshtastic/__init__.py", line 161, in _setup_meshtastic_devices
    await _setup_meshtastic_device(
        client, device_hardware_names, device_registry, entry, gateway_node, node, node_id
    )
  File "/config/custom_components/meshtastic/__init__.py", line 253, in _setup_meshtastic_device
    device_registry.async_update_device(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        d.id,
        ^^^^^
        new_connections=connections,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        via_device_id=None if via_device is None else UNDEFINED,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 982, in async_update_device
    new_values["connections"] = self._validate_connections(
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~^
        device_id, new_connections, False
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/device_registry.py", line 1070, in _validate_connections
    raise DeviceConnectionCollisionError(
        normalized_connections, existing_device
    )
homeassistant.helpers.device_registry.DeviceConnectionCollisionError: Connections {('mac', '34:b7:da:VV:VV:VV')} already registered with DeviceEntry(area_id=None, config_entries={'VVVVV'}, configuration_url=None, connections={('mac', '34:b7:da:VV:VV:VV')}, created_at=datetime.datetime(2025, 2, 4, 21, 59, 50, 352940, tzinfo=datetime.timezone.utc), disabled_by=<DeviceEntryDisabler.USER: 'user'>, entry_type=None, hw_version=None, id='VVVVV', identifiers=set(), labels=set(), manufacturer='Espressif Inc.', model=None, model_id=None, modified_at=datetime.datetime(2025, 2, 5, 21, 57, 44, 587686, tzinfo=datetime.timezone.utc), name_by_user=None, name='esp32s3-6VVVVV', primary_config_entry=None, serial_number=None, suggested_area=None, sw_version=None, via_device_id=None, is_new=False, _cache={'json_repr': b'{"area_id":null,"configuration_url":null,"config_entries":["VVVV"],"connections":[["mac","34:b7:da:VV:VV:VV"]],"created_at":1738706390.35294,"disabled_by":"user","entry_type":null,"hw_version":null,"id":"VVVVV","identifiers":[],"labels":[],"manufacturer":"Espressif Inc.","model":null,"model_id":null,"modified_at":1738792664.587686,"name_by_user":null,"name":"esp32s3-6VVVVV","primary_config_entry":null,"serial_number":null,"sw_version":null,"via_device_id":null}'})

``` 

[Log20250207 (1).log](https://github.com/user-attachments/files/18709453/Log20250207.1.log)

broglep, 2025-02-07:
Thanks for the logs. Can you please check if you are able to exclude the meshtastic device from the unify integration? 
Based on their code it looks like you you have the option to block / exclude certain devices based on mac address.

the issue is that home assistant apparently doesn't allow to device entity to share the same connection / mac address, the device that unify creates conflicts with the device that meshtastic creates. 

There might be a workaround that meshtastic device is created without the mac as connection in case of a conflict, but this could still cause issues when ordering is the other way around. 
the fundamental issue is that you are trying to (digitally) represent the very same physical device in two different ways via two different integrations in home assistant

lillpjot, 2025-02-08:
Unfortunately it seems like the disable still creates the device but not creating entities.
"Disabled devices will not be shown and entities belonging to the device will be disabled and not added to Home Assistant."

As an example on integration that share devices is the Axis integration (link below)
"https://github.com/home-assistant/core/tree/dev/homeassistant/components/axis"

Maybe there is something there how to support this.

closed 2025-02-08 by broglep

broglep, 2025-02-08:
I could reproduce it with the mikrotik integration. please update to the latest version for a fix

I have implemented the workaround, mestastic integration will fallback to only register its own connections in case of conflicts. the negative impact is only minimal, you won't see the mac address of the meshtastic gateway in the device view. 

Home assistant allows for connection conflicts in the most commonly used device registrations call, so even if meshastic registers a device earlier than other integrations, those integrations might still work

runningrunner72, 2025-02-08:
Looks like it's fixed. Thanx

lillpjot, 2025-02-08:
Excellent works for me also!
Good work and thanks for the effort.

commit 471f99e3 (2025-02-08): fix: handle device connection collision

fixes #41