meshtastic/home-assistant #43: Integration proxy for MeshSense

stefkoch09, 2025-02-08:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

It's not a problem, but would extend the use of the proxy function of the integration for use with the popular MeshSense application.

### Describe the solution you'd like

As much as the integration allows to connect to the web client, it would be very useful to connect the MeshSense appliaction, which is based on the tcp connection

### Describe alternatives you've considered

I did try to connect with MeshSense as described for the web client, but no connection could be astablished.

### Additional context

MeshSense is an Open-source development of Affirmatech: https://affirmatech.com/meshsense

broglep, 2025-02-08:
Can you elaborate a bit more about your use case of using this integration at the same time with MeshSense?
To me it feels like this integration+Home Assistant and MeshSense do have quite some overlap, and using both at the same time feels rather strange. 

So far, this integration only supports the web client http based api and not yet tcp or bluetooth interfaces that the other clients use. This proxying is still kind of experimental but would be doable to implement for tcp as well. 

I had a quick look at MeshSense, it actually does use the http based api and you are able to enter the proxy address: 
`<home assistant>:8123/meshtastic/web/<config entry id>`. You can get that address when you open the web client inside home assistant and press connect. I could verify that MeshSense does open connection and request the configuration.
But MeshSense then remains stuck in "Configuring" state, this is because MeshSense does not to respect the cookie that this integration proxy does set in order to distinguish between different connections. Further calls to retrieve messages by MeshSense do not include the cookie value, thus the proxy can't forward any messages to MeshSense. 
I'm not familiar with electron, might be that cookie supports needs to be somehow enabled. You should open a bug on MeshSense project regarding this issue.

stefkoch09, 2025-02-09:
Thanks for your quick reply and analysis!

Regarding your use case question: MeshSense gives in addition to your HA integration (which is really greaz and very useful) a very nice visualization of the mesh network. It allows easily to explose the area around you and gives even a bit of an indication to plan expanding the mesh, defing the device roles, etc. with your community.
Also, it allows to send and receive messagesand buffers them in the application log.
So, at least for me, there are a number of use cases.

I have also tried to configure it similar to your description and failed as it got stuck in the configuring state ...
Since I have no knowledge in software development, I would not even know how to describe the cookie problem in a qualifying feature or bug request at Affirmatech.

I will check. Many thanks anyway for this great integration!

TheMalkavien, 2025-03-04:
Hi,

Thanks for your work, this is very impressive, and useful !
I have kind of the same need. I use my gateway mainly with the official Android App. To provide a proxy is a really good workaround but works only when you can type a URL, not a single IP like the official app. The proxy have to handle this use case (my guess is people a using the app more than the web client, I may be wrong tho).

The alternative ofc is to use a MQTT broker (I already have this use case, and a dedicated homeassistant integration) but this is not the topic of your work.

Your integration is really promising, but for now, by using the tcp api, it broke some usages (like the official app, or the Meshsense one) - The proxy workaround just have to be more... versatile to overcome the downsides :)

Again, thanks for your work !
Regards,

broglep, 2025-03-04:
I think the official android app only supports bluetooth, or this it also support TCP or the HTTP protocol?
The official web app supports full url (or rather I would say they don't have proper input concept for he endpoint, which has nice side effect of being able to use ip, different port and & url paths) 

The actual solution would be that meshtastic firmware allows multiple connections, until that happens, we have to live with workarounds. Feel free to flag this issue upstream, I've not yet seen this happen (and you are by far not the only one facing that issue, see e.g. https://github.com/broglep/homeassistant-meshtastic/discussions/29)

Implementing bluetooth proxy would be interesting (and shouldn't be much extra work apart the potential bluetooth dongle / hardware issues that come with it), but did not yet have time yet to investigate. Supporting MQTT connection as forth way to connect to a gateway is also on the idea list for this integration but has suffers similar fate than bluetooth proxy  

I recently have invested that little spare time I have rather on the firmware side to get meshastic firmware (radio/mesh part thereof) working as a library for my ongoing & planned projects.

TheMalkavien, 2025-03-05:
Hi,

The official app supports TCP, Serial, Bluetooth. I'm using TCP as this is pretty fast, my gateway is connected by wifi, and I can connect from everywhere with my VPN. 
I'm not sure if Bluetooth is "one connection only" so I dont know if its worth implementing a proxy (I guess not)

For now I'll stay with the very incomplete (but working) Meshtastic MQTT integration, as I already have a broker.

Btw, I too have worked with the firmware code (mostly on the RP2040 compatibility) and having a multi connection TCP API should not be a lot of work. But the limited amount if memory on most modules makes this   feature a little risky.

Regards,

KmanOz, 2025-03-15:
@broglep aa quick question if I may. Based on your response above, you can the MeshSense map be visualised on HomeAssistant. I can see if configures nodes on the HA map but can I see a new node that has just appeared around me pop up on the ma without it being one of mine. Sorry if this question is stupid.

broglep, 2025-03-23:
@TheMalkavien the new pre-release version is available at https://github.com/broglep/homeassistant-meshtastic/releases/tag/v0.6.0-dev2. You can also download it via HACS, select Redownload -> Need a different version and select the version.

It now support TCP proxy as well, you can enable it in the options, would be great to get feedback if it works as desired.

broglep, 2025-03-23:
@KmanOz MeshSense is currently not compatible with this HomeAssistant integration. If you would like to see support, you can voice your desire on MeshSense project. If you have other questions, please head to [Discussions](https://github.com/broglep/homeassistant-meshtastic/discussions) and provide more details

KmanOz, 2025-03-24:
> [@KmanOz](https://github.com/KmanOz) MeshSense is currently not compatible with this HomeAssistant integration. If you would like to see support, you can voice your desire on MeshSense project. If [...]

Actually sorry I might have worded that incorrectly. I meant can nodes be viewed on the HomeAssistant map just as you can see nodes on the map in MeshSense. Not really interested in Meshsense in HA

TheMalkavien, 2025-03-24:
Hi @broglep . Thanks for the update. Since last time, as I really think your integration have potential, I scripted a quick TCP proxy in Python, just to be sure the idea works. And it does, I use your integration with my own tcpproxy, and the official app (and the CLI that I use often) is working fine with it too.

Now that you integrated this feature, I'll test it today and switch to it.

Thanks again.

TheMalkavien, 2025-03-27:
Hi, your integrated TCP proxy is working fine, I'm using it with the default android app without any problems so far. Good job !

Retidurc, 2025-05-11:
In my case the proxy is working fine when used from the android client, but a webclient or meshense is unable to connect throught it. 

The logs on HA side are not that helpfull : 
Traceback (most recent call last):
  File "/config/custom_components/meshtastic/meshtastic_tcp/server.py", line 167, in _handle_client
    await client_connection.disconnect()
  File "/config/custom_components/meshtastic/meshtastic_tcp/server.py", line 56, in disconnect
    await self._disconnect()
  File "/config/custom_components/meshtastic/meshtastic_tcp/server.py", line 63, in _disconnect
    await self._writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1067, in write
    n = self._sock.send(data)
BrokenPipeError: [Errno 32] Broken pipe

kernel-oops, 2026-11-09:
That traceback should be fixed on main now. #193 wraps the close with a deadline and falls back to `abort()`, so a dead peer's broken pipe no longer escapes the handler. The same change bounds the per-client queues and drains writes, so one client that stops reading no longer affects the others.

Still true that the real answer is the firmware allowing more than one connection, as you said earlier in this thread. Until then the proxy is what we have.