meshtastic/home-assistant #47: Location report of other nodes on the mesh

infinitytec, 2025-03-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I'd like to see where devices are on the map in Home Assistant. This would help for checking on the location of things while away and enable family members to easily check locations without having to use another platform. It would also enable looking at things from different trackers at the same time.

### Describe the solution you'd like

An ability for discovered nodes to be registered as device trackers in Home Assistant.

### Describe alternatives you've considered

n/a

### Additional context

I have two nodes. One of them is connected to Home Assistant over IP, and that one does show on the map. The other does not despite providing its location to the mesh.

broglep, 2025-03-02:
Only the devices you select in the configuration of the integration are exposed as device trackers. This is exactly designed for you use case: track location of certain nodes you are interested in.

You can always reconfigure the integration to select additional / other nodes to be tracked

infinitytec, 2025-03-02:
> Only the devices you select in the configuration of the integration are exposed as device trackers. This is exactly designed for you use case: track location of certain nodes you are interested in.
> 
> You can always reconfigure the integration to select additional / other nodes to be tracked

Well, I think I found it! Thanks for the quick implementation! 🙂

closed 2025-03-02 by infinitytec