meshtastic/home-assistant #48: No longer updating

techyporcupine, 2025-03-09:
### System Health details

## System Information

version | core-2025.2.5
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.13.4
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1575
Downloaded Repositories | 12

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 5
views | 11
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), ens18 (enabled, default, auto), tailscale0 (disabled), podman0 (disabled), veth0 (disabled), veth1 (disabled), veth2 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), ens18 (10.0.0.5/24), tailscale0 (100.64.0.5/32), podman0 (10.88.0.1/16), veth0 (), veth1 (), veth2 ()
ipv6_addresses | lo (::1/128), ens18 (redacted, redacted/64, redacted/64, redacted/64, redacted/64, redacted/128, redacted/64, redacted/64, redacted/64), tailscale0 (redacted/128, redacted/64), podman0 (redacted/64), veth0 (redacted/64), veth1 (redacted/64), veth2 (redacted/64)
announce_addresses | 10.0.0.5, redacted, redacted...

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | January 30, 2025 at 4:57 PM
-- | --
current_recorder_run | March 8, 2025 at 5:28 PM
estimated_db_size | 116.61 MiB
database_engine | sqlite
database_version | 3.47.1

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hello, I found this integration and it was working great for roughly 16 hours, but it no longer is updating any of the entities in HA. The node is still powered and recieves traceroutes, and I can't connect to it via BT, therefore the connection is still being maintained between it and the server (through a bluetooth proxy).

### Reproduction steps

1. Connect a RAK4631 running FW 2.5.20 via BT(through a proxy) to this integration
2. Observe all features working correctly
3. Wait (mine took 16 hours to have an issue)

### Debug logs

```text
2025-03-09 16:03:54.525 WARNING (MainThread) [custom_components.meshtastic] Failed to disconnect client during unload of entry
Traceback (most recent call last):
  File "/config/custom_components/meshtastic/__init__.py", line 320, in async_unload_entry
    await entry.runtime_data.client.disconnect()
  File "/config/custom_components/meshtastic/api.py", line 159, in disconnect
    await self._interface.stop()
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 315, in stop
    await self._cancel_processing_tasks()
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 334, in _cancel_processing_tasks
    await asyncio.wait(
    ...<4 lines>...
    )
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 451, in wait
    return await _wait(fs, timeout, return_when, loop)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 537, in _wait
    await waiter
asyncio.exceptions.CancelledError
```

### Diagnostics dump

Not available for this integration

KmanOz, 2025-03-10:
Can you check and see if the actual sensors are no longer being provided by the integration. After about 24Hours (I have only had it for a couple of days now) 2 or 3 times now the integration starts losing one sensor at a time and basically has to be unloaded. When I unloaded it though I notice that when I reinstalled it it remembered the state of the broken sensors. I had to to remove it, edit core_entity.registry and physically remove all sensors set up by the integration.  After reloading it worked again. Not sure what exactly is going on but will monitor and follow this for some type of resolution. Looking very promising though.

broglep, 2025-03-10:
@techyporcupine Can you provide some more logs? The snipped to attached indicates an issue while disconnecting, so this might be a hint that communication with proxy is completely broken (and not only the device).   

It would be best if you enable following logs like:

```
logger:
  default: info
  logs:
    custom_components.meshtastic: debug
    homeassistant.components.bluetooth: debug
```

in my tests, bluetooth proxy was not that stable, after some time, home assistant was not able to communicate with the proxy anymore, which then also fails the communication to the device. 

@KmanOz are you also using bluetooth proxy? Losing individual sensors at a time one after another is quite strange, are you sure that you saw that? If communication fails, none of the sensors are able to be updated, this is usually the failure mode that one can observe.

KmanOz, 2025-03-11:
No mine is an ethernet connection. Oops I missed the bluetooth thing :D Yes I can confirm that individual sensors were failing. It started the first time with the packet sensors showing a red icon saying the integration is no longer providing this sensor. I have done something different this time though. I have turned off automatic adding of nodes. I'll see how this goes. Great work though :D

techyporcupine, 2025-03-11:
@broglep Mine just had the issue again, but I forgot to enable the debug logs. Doing that now and will report back when it occurs again!

techyporcupine, 2025-03-11:
Stopped working at roughly 4:25 PM

[home-assistant_meshtastic_2025-03-11T21-43-28.860Z.log](https://github.com/user-attachments/files/19198430/home-assistant_meshtastic_2025-03-11T21-43-28.860Z.log)

[home-assistant_bluetooth_2025-03-11T21-43-25.197Z.log](https://github.com/user-attachments/files/19198433/home-assistant_bluetooth_2025-03-11T21-43-25.197Z.log)

broglep, 2025-03-11:
Thanks @techyporcupine for providing the logs. I have looked at them, at `16:36:08.191` the heartbeat failed, after that, integration is not able to communicate with the device anymore. This seems to be an issue with the bleak backend / bluetooth proxy.

Are you able to also log the bluetooth proxy serial output? Would be interesting to know if the bluetooth proxy reboots and because of that loses the connection (without HA probably noticing it). 

If you feel conformable adjusting some code locally, you can also give a try with that the following patched bluetooth file: https://github.com/broglep/homeassistant-meshtastic/blob/test/bluetooth/custom_components/meshtastic/aiomeshtastic/connection/bluetooth.py

maybe an unpair will help to properly recover in your situation

techyporcupine, 2025-03-12:
Okay, I haven't checked the logs of the Bluetooth proxy yet, but I just changed those lines of code and restarted, we'll see if it helps.

novag, 2025-03-12:
I am experiencing the same problem with the same behavior (failed heartbeat, reconnect successful, config request failed), but without a Bluetooth proxy, instead using the onboard module of the Raspberry Pi with HAOS. Restarting HomeAssistant resolves it for a few minutes/hours.

broglep, 2025-03-12:
@novag Thanks for sharing your experience, in your setup it will probably be easier to pinpoint the problem as no bluetooth proxy is involved. If you are able to log the serial output of the meshtastic device, it could be helpful to see if meshtastic device receives the config request or not.  

You can also give https://github.com/broglep/homeassistant-meshtastic/blob/test/bluetooth/custom_components/meshtastic/aiomeshtastic/connection/bluetooth.py a try and see if it helps. Other than that, I think there are currently not many other option on the integration side to address the issue. 

Anecdotally when I was developing, I also had to restart full HomeAssistant to bring bluetooth stack back in a state to be operational. Sometimes even had to delete all devices on BlueZ level.

novag, 2025-03-12:
@broglep Sure, here's the HA log and the device serial log. It looks like the device successfully confirms the reconnect and also receives the config request, but either doesn't reply or the integration doesn't receive the reply. I'll check later what the device logs on a successfully config request.

HA:
```
2025-03-12 11:31:58.696 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Reconnect requesting config did timeout, forcing next reconnect in 108 seconds
2025-03-12 11:32:06.366 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Sending heartbeat
2025-03-12 11:32:06.367 INFO (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Heartbeat failed, reconnecting
Traceback (most recent call last):
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 510, in _heartbeat_loop
    await self._connection.send_heartbeat()
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 202, in send_heartbeat
    if not await self.send_packet(p):
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 339, in send_packet
    return await self._send_packet(packet)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/bluetooth.py", line 133, in _send_packet
    raise ClientApiNotConnectedError
custom_components.meshtastic.aiomeshtastic.connection.errors.ClientApiNotConnectedError: Not connected
2025-03-12 11:33:46.699 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Starting to reconnect
2025-03-12 11:33:46.700 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect started
2025-03-12 11:33:46.700 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect: disconnecting first
2025-03-12 11:33:48.846 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect: disconnecting complete
2025-03-12 11:33:50.671 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect completed
2025-03-12 11:33:50.671 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Reconnect connection succeeded, requesting config
2025-03-12 11:34:50.673 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Reconnect requesting config did timeout, forcing next reconnect in 125 seconds
```

Device:
```
INFO | ??:??:?? 185 BLE Disconnected, reason = 0x13
DEBUG | ??:??:?? 185 PhoneAPI::close()
INFO | ??:??:?? 186 BLE Connected to
INFO | ??:??:?? 186 BLE connection secured
INFO | ??:??:?? 187 toRadioWriteCb data 0x2001e04a, len 6
DEBUG | ??:??:?? 187 New ToRadio packet
INFO | ??:??:?? 187 Client wants config, nonce=2631502170
DEBUG | ??:??:?? 187 Got 5 files in manifest
INFO | ??:??:?? 187 Start API client config
DEBUG | ??:??:?? 205 [Power] Battery: usbPower=1, isCharging=1, batMv=2698, batPct=0
DEBUG | ??:??:?? 225 [Power] Battery: usbPower=1, isCharging=1, batMv=2705, batPct=0
DEBUG | ??:??:?? 245 [Power] Battery: usbPower=1, isCharging=1, batMv=2715, batPct=0
```

broglep, 2025-03-12:
@novag great, thanks. When meshtastic node actually send out all config, it can't be missed in the serial log, it produces a lot of log entries (currently don't have a sample at hand, if you have decent sized node db, it can easily be 50+ log lines). 
So it looks like for some reasons the meshtastic node fails to send out all the data. The tear down on meshstatic side looks fine on first sight (as firmware doesn't allow multiple connection that is sometimes an issue).

Can you share which meshtastic firmware version you are using?

broglep, 2025-03-12:
looking at the firmware code, it would also be interesting to know if you all that are experiencing this are using NRF52 or ESP32 based devices

novag, 2025-03-12:
@broglep Indeed, it looks like it didn't send the config. I am on version 2.5.20.4c97351 on a RAK4631 (NRF52).

broglep, 2025-03-12:
Ok thanks for the info, will try to see if I have a spare RAK4631 available to try to reproduce. In the meantime you can give https://github.com/broglep/homeassistant-meshtastic/compare/main...test/bluetooth a try to rule out issue on the integration side (based on short glance on the firmware code, if there is no read from bluetooth, the device will not proceed with sending out the config)

techyporcupine, 2025-03-12:
I am also on a RAK4631 (nRF52), running 2.5.20 as well.

KmanOz, 2025-03-12:
For the record I also have a RAK4631. It seems the integration has now stabilised and is working correctly. All I did was stop to automatically add nodes as they appear.

novag, 2025-03-12:
@broglep Log with your bluetooth patch:

```
2025-03-12 13:56:30.252 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] _packet_stream: No more packets available, waiting for notification
2025-03-12 13:58:50.824 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Sending heartbeat
2025-03-12 13:58:50.827 INFO (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Heartbeat failed, reconnecting
Traceback (most recent call last):
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 507, in _heartbeat_loop
    await self.request_connection_status()
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 855, in request_connection_status
    response = await self.send_admin_message_await_response(node=node, message=admin_message, expect_response=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 1013, in send_admin_message_await_response
    return await self._send_message_await_response(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 979, in _send_message_await_response
    return next(iter(done)).result()
           ~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 275, in send_mesh_packet
    return await self._send_await_response(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        to_radio, want_response=want_response, ack_callback=ack_callback, response_callback=response_callback
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 289, in _send_await_response
    async for from_radio in self.listen(on_start=self.send_packet(to_radio)):
    ...<31 lines>...
        return ack_packet
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 110, in listen
    raise ClientApiNotConnectedError
custom_components.meshtastic.aiomeshtastic.connection.errors.ClientApiNotConnectedError: Not connected
2025-03-12 13:58:50.867 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Starting to reconnect
2025-03-12 13:58:50.867 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect started
2025-03-12 13:58:50.867 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect: disconnecting first
2025-03-12 13:58:50.868 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect: disconnecting complete
2025-03-12 13:58:52.141 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect completed
2025-03-12 13:58:52.142 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Reconnect connection succeeded, requesting config
2025-03-12 13:59:52.146 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Reconnect requesting config did timeout, forcing next reconnect in 1 seconds
```

broglep, 2025-03-12:
Thanks @novag, that is very valuable help. That the bleak client seems to have no backend anymore, seems to be very strange. But not sure if it is tied to this the actual problem. I've added another commit with some more debug output. 

Could not yet find RAK4631, so have not yet a setup to test on my own

techyporcupine, 2025-03-12:
Random info that may or may not be helpful. Now that I think back, I used to have this node connected to my phone most of the time, and I would notice that occasionally the app would just disconnect from the node, and not reconnect unless I manually selected "None" in the app pairing menu, and then selected my node again.

Maybe this isn't an issue with the integration at all, and rather the RAK devices or the firmware?

broglep, 2025-03-12:
Yes, it very well possible that this is a firmware or bluetooth stack issue (on meshtastic device or HA side). I now have a test setup running as well and will see if I also get into such a state. Additional meshtastic node serial logs in connection with the debugs logs from the [test/bluetooth](https://github.com/broglep/homeassistant-meshtastic/blob/test/bluetooth/custom_components/meshtastic/aiomeshtastic/connection/bluetooth.py) branch might be helpful if anybody else gets into that state.

techyporcupine, 2025-03-12:
Had the issue even after changing the lines you said. For some reason it downloaded the logs as .bin files. I'll try to upload them asap.

techyporcupine, 2025-03-12:
```
2025-03-12 12:17:16.497 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.MeshInterface] Starting to reconnect
2025-03-12 12:17:16.497 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect started
2025-03-12 12:17:16.497 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect: disconnecting first
2025-03-12 12:17:16.497 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Unpairing failed
Traceback (most recent call last):
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 375, in _heartbeat_loop
    await self.request_connection_status()
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 721, in request_connection_status
    response = await self.send_admin_message_await_response(node=node, message=admin_message, expect_response=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 879, in send_admin_message_await_response
    return await self._send_message_await_response(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 845, in _send_message_await_response
    return next(iter(done)).result()
           ~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 275, in send_mesh_packet
    return await self._send_await_response(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        to_radio, want_response=want_response, ack_callback=ack_callback, response_callback=response_callback
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 289, in _send_await_response
    async for from_radio in self.listen(on_start=self.send_packet(to_radio)):
    ...<31 lines>...
        return ack_packet
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 110, in listen
    raise ClientApiNotConnectedError
custom_components.meshtastic.aiomeshtastic.connection.errors.ClientApiNotConnectedError: Not connected

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/bluetooth.py", line 86, in _disconnect
    await self._bleak_client.unpair()
  File "/usr/local/lib/python3.13/site-packages/bleak/__init__.py", line 653, in unpair
    return await self._backend.unpair()
                 ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'unpair'
2025-03-12 12:17:16.499 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Reconnect: disconnecting complete
2025-03-12 12:17:16.499 DEBUG (MainThread) [custom_components.meshtastic.aiomeshtastic.connection.BluetoothConnection] Failed to reconnect
Traceback (most recent call last):
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 375, in _heartbeat_loop
    await self.request_connection_status()
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 721, in request_connection_status
    response = await self.send_admin_message_await_response(node=node, message=admin_message, expect_response=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 879, in send_admin_message_await_response
    return await self._send_message_await_response(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/config/custom_components/meshtastic/aiomeshtastic/interface.py", line 845, in _send_message_await_response
    return next(iter(done)).result()
           ~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 275, in send_mesh_packet
    return await self._send_await_response(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        to_radio, want_response=want_response, ack_callback=ack_callback, response_callback=response_callback
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 289, in _send_await_response
    async for from_radio in self.listen(on_start=self.send_packet(to_radio)):
[... 125 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/__init__.py", line 456, in reconnect
    await self._connect()
  File "/config/custom_components/meshtastic/aiomeshtastic/connection/bluetooth.py", line 57, in _connect
    await self._bleak_client.connect()
  File "/usr/local/lib/python3.13/site-packages/habluetooth/wrappers.py", line 286, in connect
    wrapped_backend = self._async_get_best_available_backend_and_device(manager)
[... cut ...]
2025-03-12 12:19:00.569 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139974926391696] Initialized trigger
2025-03-12 12:19:00.589 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139974926391696] Initialized trigger
2025-03-12 12:19:00.590 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139974926391696] Initialized trigger
2025-03-12 12:19:00.590 INFO (MainThread) [homeassistant.components.websocket_api.http.connection] [139974926391696] Initialized trigger
```

broglep, 2025-03-12:
@techyporcupine in your case, is seems to be a bluetooth proxy issue: `No backend with an available connection slot that can reach address C4:02:33:2F:FD:62 was found`

Make sure that your phone does not try to connect to the node, otherwise you get this as well.

techyporcupine, 2025-03-12:
Yeah, I thought it could be my phone but I wasn't even home during the past couple issues. Do you know if I can blacklist that Mac address for use in the Bluetooth proxy, so it's forced to use the Bluetooth on my server?

techyporcupine, 2025-03-12:
Hmm actually this latest issue doesn't quite seem like one with your integration. The node has just entirely stopped responding to any communication.

broglep, 2025-03-13:
I could observe it once after one day (using BLE proxy), in that case it seemed to be issue with bluetooth notification. Still unclear if on HA or device side. There might be a workaround possible by the integration to not only relying on bluetooth notifications but after some time try an active read

Stuck Case: 
```
INFO  | 13:50:30 78927 toRadioWriteCb data 0x2001e57a, len 6
DEBUG | 13:50:30 78927 New ToRadio packet
INFO  | 13:50:30 78927 Client wants config, nonce=3763004891
DEBUG | 13:50:30 78927 Got 6 files in manifest
INFO  | 13:50:30 78927 Start API client config
[stuck]
```

Working Case:
```
INFO  | 13:57:47 79364 toRadioWriteCb data 0x2001e57a, len 6
DEBUG | 13:57:47 79364 New ToRadio packet
INFO  | 13:57:47 79364 Client wants config, nonce=1503766717
DEBUG | 13:57:48 79364 Got 6 files in manifest
INFO  | 13:57:48 79364 Start API client config
INFO  | 13:57:48 79364 CCCD Updated: 1
INFO  | 13:57:48 79364 Notify/Indicate enabled
DEBUG | 13:57:48 79364 FromRadio=STATE_SEND_MY_INFO
INFO  | 13:57:48 79364 getFromRadio=STATE_SEND_UIDATA
[...]
```

novag, 2025-03-13:
@broglep Your assumption that it is related to the bluetooth notification prompted me to do a simple test. I replaced line 131 in bluetooth.py (https://github.com/broglep/homeassistant-meshtastic/blob/test/bluetooth/custom_components/meshtastic/aiomeshtastic/connection/bluetooth.py#L131) with the following snippet:

```python
try:                                                                                                   
    await asyncio.wait_for(packet_num_queue.get(), timeout=10)            
except asyncio.TimeoutError:                                                                      
    self._logger.debug("_packet_stream: packet wait timeout")
```

I can't say yet whether this actually prevents the heartbeat timeout, but it does have a different effect. The WebUI is now working. Previously, when connecting the Meshtastic WebUI it would sent an initial packet to the Meshtastic device but never received a response, which led to my WebUI not displaying any data (all fields showing "UNK"). With the patch, the WebUI suddenly works. The response did not trigger a notify, but now the response is correctly read on timeout. I did not look into it any further yet, so I can't say whether it's related to the actual problem.

novag, 2025-03-14:
@broglep So yes this seems to prevent the connection from getting stuck. Even after the BT connection drops it now automatically reconnects.

broglep, 2025-03-14:
I have a similar patch running now as well and looks like it helps. Having an active bluetooth gatt read after a certain timeout also seems to help bleak to detect that something went wrong. I got the following exception after notification stopped working: 

```
bleak.exc.BleakError: Peripheral E6:31:D4:XX:XX:XX changed connection status while waiting for BluetoothGATTReadResponse, BluetoothGATTErrorResponse: Unknown error (0)
```

broglep, 2025-03-23:
A new pre-release version is available at https://github.com/broglep/homeassistant-meshtastic/releases/tag/v0.6.0-dev2
You can also download it via HACS. Select Redownload -> Need a different version and select the version. 

Timeout is now less aggressive 5 minutes, if no bluetooth notification are received 3 consecutive times, an attempt to restart bluetooth notifications is made. With that it should be pretty reliable now. 

As it takes quite a while for me to reproduce, it would be great if everybody that is affected by this issues could give the new version a try and report if connection is now stable

novag, 2025-03-23:
@broglep Thanks! The web ui still doesn't work. However, with a timeout of 10 seconds it works. If the problems are really connected you might be able to test it with the web ui (or the new TCP proxy).

closed 2025-03-26 by broglep

techyporcupine, 2025-04-02:
@broglep Just to give you a status update after testing this for a while, the issue seems to be resolved! Even through the same BT proxy, it has been holding stable and functioning well!

commit bb2cd4b3 (2025-03-23): fix: recover from cases where bluetooth notification stop working

fixes #48
commit 39036aa8 (2025-03-23): fix: recover from cases where bluetooth notification stop working

fixes #48
commit 753f0123 (2025-03-23): fix: recover from cases where bluetooth notification stop working

fixes #48
commit 992b6f00 (2025-03-26): fix: recover from cases where bluetooth notification stop working

fixes #48