meshtastic/home-assistant #50: Web Client still accessible after disabling

ov1d1u, 2025-03-11:
### System Health details

## System Information

version | core-2025.3.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.62-haos-raspi
arch | aarch64
timezone | Europe/Bucharest
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4979
Installed Version | 2.0.5
Stage | running
Available Repositories | 1670
Downloaded Repositories | 63

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | September 17, 2025 at 03:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | eu-central-1-11.ui.nabu.casa
certificate_status | ready
instance_id | 2de5d6f10db943018dfcafc0b7869525
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.2
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 458.4 GB
disk_used | 28.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Cloudflared (5.2.10), Dnsmasq (1.8.1), File editor (5.8.0), Mosquitto broker (6.5.0), Network UPS Tools (0.14.1), Node-RED (19.0.2), Rclone Backup (3.3.4), Samba share (12.5.0), chrony (5.1.0), Terminal & SSH (9.16.0), Zigbee2MQTT Edge (edge), Piper (1.5.2), openWakeWord (1.10.0), Whisper (2.4.0), Earthquake Alert (0.44), Zigbee2MQTT (2.1.3-1), ESPHome Device Builder (2025.2.2), Studio Code Server (5.18.3), MPD (1.7.5), OpenThread Border Router (2.13.0), Matter Server (7.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 29
views | 27
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth03db589 (disabled), vethd4922bc (disabled), veth4690f4c (disabled), veth6d736e8 (disabled), veth609ba1e (disabled), veth279dee7 (disabled), veth9778c3e (disabled), vethb02e8ed (disabled), vethc82e9cd (disabled), wpan0 (disabled), veth297f234 (disabled), veth10ce725 (disabled), veth487b5a5 (disabled), veth47bfa48 (disabled), vethcf6d8f7 (disabled), veth22ce131 (disabled), veth4f00bb5 (disabled), vethfe5a465 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.0.100/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth03db589 (), vethd4922bc (), veth4690f4c (), veth6d736e8 (), veth609ba1e (), veth279dee7 (), veth9778c3e (), vethb02e8ed (), vethc82e9cd (), wpan0 (), veth297f234 (), veth10ce725 (), veth487b5a5 (), veth47bfa48 (), vethcf6d8f7 (), veth22ce131 (), veth4f00bb5 (), vethfe5a465 ()
ipv6_addresses | lo (::1/128), end0 (fdb0:732c:a83f:146f:e316:f1ff:c152:edb1/64, fe80::b10f:87b5:e2d4:c256/64), docker0 (fe80::42:48ff:fee7:71fc/64), hassio (fe80::42:bff:fe0d:c436/64), veth03db589 (fe80::2c00:6fff:fef2:d3d1/64), vethd4922bc (fe80::70df:89ff:fe42:b223/64), veth4690f4c (fe80::780f:f4ff:fed6:59d2/64), veth6d736e8 (fe80::48ac:8dff:fe3d:d18/64), veth609ba1e (fe80::34ea:6ff:fe15:d423/64), veth279dee7 (fe80::f05a:cdff:fe65:1461/64), veth9778c3e (fe80::f4f8:8cff:feaa:b857/64), vethb02e8ed (fe80::b0f7:b5ff:fe1b:4721/64), vethc82e9cd (fe80::e01e:47ff:fe99:5700/64), wpan0 (fdc2:edb8:dc8b:b8d8:0:ff:fe00:fc11/64, fde5:b733:32ca:1:e11a:a141:2704:8b69/64, fdc2:edb8:dc8b:b8d8:0:ff:fe00:fc10/64, fdc2:edb8:dc8b:b8d8:0:ff:fe00:fc38/64, fdc2:edb8:dc8b:b8d8:0:ff:fe00:fc00/64, fdc2:edb8:dc8b:b8d8:0:ff:fe00:400/64, fdc2:edb8:dc8b:b8d8:d5e6:e8f5:58b3:c070/64, fe80::c3d:c10c:99d:a2a4/64), veth297f234 (fe80::4053:7cff:fe84:9c73/64), veth10ce725 (fe80::f42e:64ff:febe:92c3/64), veth487b5a5 (fe80::e4d6:69ff:fe9f:b263/64), veth47bfa48 (fe80::c8e3:87ff:fea5:f775/64), vethcf6d8f7 (fe80::58fa:8ff:fe6d:e2ee/64), veth22ce131 (fe80::d886:8dff:fe23:f3e8/64), veth4f00bb5 (fe80::182d:6fff:feff:d283/64), vethfe5a465 (fe80::8083:cfff:fe07:4ab1/64)
announce_addresses | 192.168.0.100, fdb0:732c:a83f:146f:e316:f1ff:c152:edb1, fe80::b10f:87b5:e2d4:c256

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 22, 2025 at 13:13
-- | --
current_recorder_run | March 11, 2025 at 10:37
estimated_db_size | 547.55 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

After I disable Web Client in the config entry's settings, I still see the icon in the sidebar and I'm able to open the web client.

### Reproduction steps

1. Configure Meshtastic integration with web client enabled.
2. Go to Settings -> Devices & Services -> Meshtastic
3. Disable web client

### Debug logs

```text
.
```

broglep, 2025-03-12:
Did you try to actually connect after having opened the web client? This should not be possible.
You should get "Web client not enabled for gateway"

Can add some support that the panel is unloaded when no gateway has web client active. 
On the panel itself it is not easily possible to filter out gateways that don't have it enabled, couldn't find an easy way to access that information (adding an entity for only that felt overkill). The "Open" button is only greyed out when gateway is not available/connected.

ov1d1u, 2025-03-13:
Yes, after I disable the web client I'm still able to connect (event after a hard refresh - Shift + F5). I can see things like battery level, firmware version, nodes and even communicate.

https://github.com/user-attachments/assets/c4f5a324-256a-44c2-b829-98c4400199ca

broglep, 2025-03-13:
thanks for screencast, that is strange, behaves as the config was not really saved/applied. 
Can you have a look at `<ha config directory>/.storage/core.config_entries` file? 
It should contain `"web_client":{"enable":true}}` or `"web_client":{"enable":fase}}` when enabled or disabled

Home Assistant debug logs might also show some hints in case something did not work when applying the changed settings

ov1d1u, 2025-03-13:
It's `"web_client":{"enable":false}}`.

I found out that if I disable the Web UI but I try to force the URL to `http://192.168.0.100:8123/meshtastic/web/index.html?path=/meshtastic/web/01JP4KF91VEXKGGX....`, it loads the Web UI which is also completely functional.

broglep, 2025-03-13:
I had another look, the index.html is served without check, and `toradio` and `fromradio` api calls also don't have a check.
There might be also some browser caching in play, in the first step the url `http://192.168.0.100:8123/meshtastic/web/01JP4KF91VEXKGGX....` is being loaded (which has the check) and then redirects to `http://192.168.0.100:8123/meshtastic/web/index.html?path=/meshtastic/web/01JP4KF91VEXKGGX....`

Unfortunately it seems to be not possible to unregister rest resources in home assistant, so when web client was enabled once, they will remain available. Will have to add the check also to `toradio`and `fromradio` api calls (for the index.html it is not really possible). Downside is that it will still look like it would work, but when you connect via web ui, connection will not establish successfully

commit 59f2eb63 (2025-03-23): feat: unload meshtastic panel when not needed anymore

fixes #50
commit 7da709b7 (2025-03-23): fix: check if web client is enabled for meshstatic api endpoints

fixes #50
commit e7b710a1 (2025-03-23): fix: check if web client is enabled for meshstatic api endpoints

fixes #50
commit 656d4d79 (2025-03-23): fix: check if web client is enabled for meshstatic api endpoints

fixes #50
closed 2025-03-26 by broglep