meshtastic/home-assistant #53: does not send a message

tcmouratec, 2025-03-21:
### System Health details

even connected and accessing by HA the client web page opens normal the others in and devices are recognized but it does not send message neither eh channel nor in the private only receives message in the private but does not respond 

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

even connected and accessing by HA the client web page opens normal the others in and devices are recognized but it does not send message neither eh channel nor in the private only receives message in the private but does not respond 

### Reproduction steps

1.
2.even connected and accessing by HA the client web page opens normal the others in and devices are recognized but it does not send message neither eh channel nor in the private only receives message in the private but does not respond 
3.
...

### Debug logs

```text
even connected and accessing by HA the client web page opens normal the others in and devices are recognized but it does not send message neither eh channel nor in the private only receives message in the private but does not respond
```

### Diagnostics dump

even connected and accessing by HA the client web page opens normal the others in and devices are recognized but it does not send message neither eh channel nor in the private only receives message in the private but does not respond

broglep, 2025-03-23:
Please describe your issue more in detail, this report is too vague and not actionable.

tcmouratec, 2025-03-25:
Thanks resolved f5 f5 sucess

closed 2025-03-26 by broglep