meshtastic/home-assistant #54: document how to (on our own) update the web front end

pleasantone, 2025-03-22:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The web front end that’s in GitHub here is old and in some ways is a “compiled binary blob” even though it is JavaScript.

Would it be possible/easy to either: document how you built it and what local changes you made if any from the source repo for that project?

### Describe the solution you'd like

Even cooler is, if you were to reference to the source repo as a git submodule and built the compiled objects as part of the “build process” for this repo. That would make it trivial to keep that “third-party software” up to date.

I’d be willing to submit a pull request myself, if you were to be able to easily document what your current build process is for the web front end, but I really don’t wanna have to do a bunch of three-way diff on “de-compiled” and merged code if I could avoid it.

### Describe alternatives you've considered

Just document the procedure for creating the blobs in this repo, it could even just be a shell script.

### Additional context

None

broglep, 2025-03-23:
It is still experimental feature, so did not invest much time in getting it fully integrated.

The changes are "documented" here: https://github.com/meshtastic/web/compare/master...broglep:meshtastic-web:homeassistant

broglep, 2025-03-23:
here has been changes on how the frontend is being built, I'll have to investigate how to update

broglep, 2025-03-23:
Please check the latest pre-release which contains the most recent version of meshtastic web. There was no official release since Nov 2.5.4, so the latest master as of now is being used. It also required some small changes on the backend side because frontend now used an additional endpoint.  

Should they release a stable version, this integration will likely update to that version. Currently I don't have any plans to remain bleeding edge with the web frontend. 

But if you'd like contribute and integrate the building of the assets into this repo, it would be greatly appreciated and we could track upstream more closely.

pleasantone, 2025-04-24:
Just following up here, I've been quiet--got slammed with work, but it is my intention to get around to it. I see it's a bit more complicated than I had hoped, but I still think we can bring the web client in as a submodule, patch it automatically, and build.

Even better, since you've gotten the blessing of the MT project, perhaps we could get your changes integrated and we can just build with a switch?

broglep, 2025-07-06:
I recently tried to update to 2.7.0, but could not fix it within reasonable time, so I had to revert to 2.6.4 and do some fixes there. 
web project seems still too unstable to attempt building something automated. 

There are various other issues with the web frontend not directly related to the home assistant integration, e.g. path handling for various resources is kind of broken (only works when application is served from root path)

Will close this now, but if you find time and get a working solution, I'd be more than happy to accept PR. I've created a issue to track this: #83

closed 2025-07-06 by broglep