meshtastic/home-assistant #56: Cannot send message

sylvain-hazera, 2025-03-26:
### System Health details

Hello,
I am trying to make automation to send message to node or channel.
when trying to create automation to send message, it does not work.

I have an error : 
Erreur : Action notify.mesh_channel_fr_balise not found

here my first automation for mesh, I just want to try to send a message.

alias: Mesh notfy meteo
description: ""
triggers:
  - trigger: state
    entity_id:
      - sensor.33_weather_alert
conditions: []
actions:
  - action: notify.mesh_channel_fr_balise
    metadata: {}
    data:
      message: Meteo
mode: single

thanks

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

when click : execute, nothing happens

### Reproduction steps

1.create automation
2.right click
3.send
...

### Debug logs

```text
Erreur : Action notify.mesh_channel_fr_balise not found
```

sylvain-hazera, 2025-03-26:
note : I have the same error if I choose a node instead of channel.

thanks.

broglep, 2025-03-26:
You are using it wrong, `notify.mesh_channel_fr_balise` is an entity_id, not an action, please check the official documentation: https://www.home-assistant.io/integrations/notify/#example-with-the-entity-platform-notify-action

closed 2025-03-26 by broglep