meshtastic/home-assistant #57: only default channel has entity : notify.mesh_node_channel created, secondary channel does not have

sylvain-hazera, 2025-03-26:
### System Health details

hello, everything is here

only default channel has entity : notify.mesh_node_channel created, secondary channel does not have

thanks

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

only default channel has entity : notify.mesh_node_channel created, secondary channel does not have

### Reproduction steps

1.when creating first node
2.only default channel has entity : notify.mesh_node_channel created, secondary channel does not have
3.
...

### Debug logs

```text
no logs identify for this
```

broglep, 2025-04-13:
Can you please check your meshtastic node config? The integration creates entities for all available channels that are not disabled (channel role must not be "DISABLED")

broglep, 2025-07-06:
Feel free to re-open if still an issue

closed 2025-07-06 by broglep