meshtastic/home-assistant #58: Meshtastic Web Clients broken in v0.6.0

timniklas, 2025-03-27:
### System Health details

## System Information

version | core-2025.3.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.73-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4994
Installed Version | 2.0.5
Stage | running
Available Repositories | 1617
Downloaded Repositories | 23

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 1. Januar 2018 um 01:00
relayer_connected | false
relayer_region | null
remote_enabled | false
remote_connected | false
alexa_enabled | false
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | null
certificate_status | null
instance_id | 1d58e09f8a334da7b1c2fdb729cd7036
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 109.3 GB
disk_used | 33.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (20.0.2), File editor (5.8.0), Log Viewer (0.17.1), ESPHome Device Builder (2025.3.2), rtl_433 MQTT Auto Discovery (0.9.0), Let's Encrypt (5.4.3), NGINX Home Assistant SSL proxy (3.11.1), Studio Code Server (5.18.3), UniFi Network Application (4.0.1), Home-Assistant-Matter-Hub (3.0.0-alpha.76), Mosquitto broker (6.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 4
views | 14
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp1s0 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth0442d4f (disabled), vethb4e9bc9 (disabled), veth2342392 (disabled), veth7109d9e (disabled), vethe2fb5ac (disabled), veth3b4b2c5 (disabled), vethcc56b30 (disabled), vetheb1c0c7 (disabled), veth734de84 (disabled), vetha414b49 (disabled), veth9222b0b (disabled), veth4af71fd (disabled), vethed1d99c (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp1s0 (NOPE/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth0442d4f (), vethb4e9bc9 (), veth2342392 (), veth7109d9e (), vethe2fb5ac (), veth3b4b2c5 (), vethcc56b30 (), vetheb1c0c7 (), veth734de84 (), vetha414b49 (), veth9222b0b (), veth4af71fd (), vethed1d99c ()
ipv6_addresses | lo (::1/128), enp1s0 (fe80::3657:e9c8:e2e8:9a07/64), docker0 (fe80::42:12ff:fe5e:6f9e/64), hassio (fe80::42:7eff:feea:690c/64), veth0442d4f (fe80::44d0:83ff:fee4:323b/64), vethb4e9bc9 (fe80::c52:ff:fe23:60dc/64), veth2342392 (fe80::bc23:87ff:fe20:b9ac/64), veth7109d9e (fe80::246c:2dff:fe08:a9b4/64), vethe2fb5ac (fe80::b0f1:64ff:fe81:ae1b/64), veth3b4b2c5 (fe80::54d2:37ff:fe6b:dcc4/64), vethcc56b30 (fe80::906a:faff:fe7b:36c1/64), vetheb1c0c7 (fe80::c4c2:59ff:fe16:3a2c/64), veth734de84 (fe80::5c11:54ff:fe3a:615/64), vetha414b49 (fe80::c440:2cff:fe7c:83f5/64), veth9222b0b (fe80::2cc9:beff:fe63:91f7/64), veth4af71fd (fe80::ac9c:e9ff:fe3e:e029/64), vethed1d99c (fe80::d4c8:eff:fee9:73d4/64)
announce_addresses | NOPE, fe80::3657:e9c8:e2e8:9a07

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 15. März 2025 um 21:32
-- | --
current_recorder_run | 27. März 2025 um 10:51
estimated_db_size | 553.94 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The web client is not working anymore since the update, I think the URL i get redirected to is wrong.

[image]

### Reproduction steps

1. Click on Meshtastic in the sidebar
2. Click "open"
3.  gets redirected to "/meshtastic/web/gateway" -> id missing?
4. 404 Error

### Debug logs

```text
how do i upload??
```

broglep, 2025-03-27:
I was not able to reproduce this, the URL should contain the second part of entity_id of the gateway, so something like
 `http://<home-assistant>/meshtastic/web/gateway_<shortName>` when your gateway entity_id is `meshtastic.gateway_<shortName>`. Can you check & share your entities?

To also rule out some browser caching issues, you can try an incognito browser session? 

It could be that it just needs a bit patience before it starts working. The api endpoints in home assistant (that do the forwarding & serving) are registered only after the meshtastic config entries are fully loaded, when your meshtastic device is slow, it can take 30 - 60 seconds. 
You can enable debugging logging for the meshtastic integration if the issue persists, so that I can check what is going on.

timniklas, 2025-03-27:
Unfortunately, I was able to recreate it with a friend on the identical setup:
- RAK 4631 via ESPHome Proxy
- Latest Home Assistant version
- Latest Meshtastic integration

The link to which you are redirected is also /meshtastic/web/gateway (without ID)

The TCP proxy does not work either, in both cases.

Caches have been deleted (F12 + disable cache and Ctrl+F5), this should not be the problem.

We both use smilies as short names, could that be a problem?
I can't access the path /meshtastic/web/gateway_📡

timniklas, 2025-03-27:
Looking at the entity ids the smiley seems to be the problem

[image]

timniklas, 2025-03-27:
Working just fine after changing the shortName to a text based one

asjmcguire, 2025-04-29:
Ah, no. This used to work. 
I too cannot access the web ui now.
My stations shortname is 🌥️
even if I try to access gateway_🌥️

If tries to access /meshtastic/web/gateway_🌥%EF%B8%8F which results in 404

broglep, 2025-04-29:
> If tries to access /meshtastic/web/gateway_🌥%EF%B8%8F which results in 404

That short name seems to have some extra unicode data (VARIATION SELECTOR-16 in this case). How was the short name configured? Could it be related to the firmware version or the applications used to configure that causes this. Would be interesting to have the node setting (in protobuf binary form).

broglep, 2025-07-06:
the issue is how emojis are treated by home assistant inside suggest object id. A `gateway 🌥️` will be converted to just `gateway` and not as the code expects to `gateway_<shortname>`. This causes the frontend serving code to ignore it. 

I have a fix that will resort to long name in such cases. A suffix is required, otherwise you get issues as in #17  (on conflicts when you have to gateways with emojis)

commit b1bca937 (2025-07-06): fix: object id when short name is emoji

fixes #58
closed 2026-11-09 by jamesarich