meshtastic/home-assistant #59: Feature Request: Show messages sent from HA in logbook

techyporcupine, 2025-04-01:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I dislike it when I am having a conversation on the mesh, and I can't see the messages that I sent from HA without checking a different node of mine.

### Describe the solution you'd like

I would like to have the messages from the node connected to HA show up either in the primary channel logbook entry, or in their own logbook entry, just some way that they can be seen from HA.

### Describe alternatives you've considered

Going to a different node to see what was sent, but that node doesn't always have all the messages as it's a mobile node, not a base node (like the HA one is).

### Additional context

Don't think there is any other context. Let me know if any other info is needed!

Blackbeardaa, 2025-04-03:
i see every message but the ones i send with the notify. in the log book for any channel I use it on. i can see what other nodes say and what i say from a node in log but not anything sent from this integration

techyporcupine, 2025-04-03:
Correct, that is why this is a feature request.

broglep, 2025-04-13:
Good suggestion, I think it makes sense to also log outgoing messages to the log book. Details still TBD, I'll have to look into it.

techyporcupine, 2025-04-13:
Perfect, thank you!

commit fd31db44 (2025-07-06): feat: record outgoing messages to logbook

fixes #59
closed 2026-11-09 by jamesarich