meshtastic/home-assistant #62: Urgent Security Issue

Paxy, 2025-04-13:
### System Health details

## System Information

version | core-2025.4.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.12.20-haos
arch | x86_64
timezone | Europe/Belgrade
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1656
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.04.0
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 23.9 GB
disk_used | 20.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Samba share (12.5.0), Mosquitto broker (6.5.0), ESPHome Device Builder (2025.3.3), Matter Server (7.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 3
views | 17
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp0s18 (enabled, default, auto), docker0 (disabled), hassio (disabled), vethf9977d8 (disabled), vetha3d6a9f (disabled), veth2a9ff26 (disabled), vethd82cf44 (disabled), veth6fd7d02 (disabled), vethc514fcc (disabled), veth736658d (disabled), veth2bccf6f (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp0s18 

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 22, 2025 at 07:57
-- | --
current_recorder_run | April 13, 2025 at 13:44
estimated_db_size | 6391.82 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

It works great. Great job with this integration.
BUT!
Note that with current implementation of Meshtastic client, if HomeAssistant is publicly accessable, everyone can get into your Meshtastic node using URL they see in for example screenshot:
https://**********:8123/meshtastic/web/index.html?path=/meshtastic/web/01JRQDG8ERVG2Y2BVS4FMGS***

Please add some kind of PIN or any other authentication.

### Reproduction steps

1. Install addon
2. Go the the Meshtastic client from side bar
3. Open URL in Incognito mode

### Debug logs

```text
No debug logs available.
```

broglep, 2025-04-13:
This is a known issue / risk, because of that it is not enabled by default (see https://github.com/broglep/homeassistant-meshtastic/issues/26#issuecomment-2599836208)

I though that security implication is also documented in the UI when enabling it, but it seems like this did not make it into the main version. Apart from making this security implication more visible to the end user, there is not much that can be done about it on a technical level. Web UI does not work when having an additional authentication layer, and the meshtastic protocol is inherently unauthenticated (with only bluetooth being the exception when bluetooth PIN is being used).

There is small protection that you need to know HA config entry id, something that should not accessible without authentication. 

Home assistant should usually be run on a local network by kind of tech savvy & security/privacy aware people, so with proper documentation about that fact that everybody in the network can access that gateway, this should acceptable that everybody can make their own risk assessment and security decision

Paxy, 2025-04-13:
Can I suggest a solution for integration so "/meshtastic/web/index.html" is served only to private IPs, or even only on pre defined IP list. This could be added in configuration option?

broglep, 2025-04-13:
The actual security issue can only be solved when adding protection to the API endpoints `/web/{config_entry_id}/api/v1/toradio` and `/web/{config_entry_id}/api/v1/fromradio`, which is not really possible without substantial changes to the meshtastic communication protocol and/or the web frontend, something that is completely out of scope as port of this HA integration. Note that this not only applies to the web ui, but also the TCP proxy. 

I don't believe it does make much sense to implement some sort of perimeter security into this integration. If you expose home assistant to the public internet (hopefully not direct but with behind firewall or a reverse proxy), you should have the capabilities already on that infrastructure level to protect those web ui endpoint as necessary (if you trust perimeter security)  

Adding some sort of PIN or similar protection to just the UI gives a false security impression.

Paxy, 2025-04-13:
I know that Mashtastic client is out of scope for this project. But I can't see the reason you can't implement basic filtration on integration endpoin level "/meshtastic/web/". Simply does not allow that page to be served from public IP would be some solution. There are HASS implementation that just expose port 8123 to the public without reverse proxy.

broglep, 2025-04-13:
I know it seems tempting at first sight, but there are actually quite some reasons against implementing something like this.
I personally wouldn't trust a custom component doing such a filtration correct, and I also do not trust myself (enough without additional review) that I can implement a secure basic filtration that work under all circumstances (with any adventurous network setup users might come up) and that lives up to the security promise it appears to give. 

Just some questions that are not that easy to answer if you look at them closely:
- How do you define as a "public IP"?
- Which source IP should be chosen? (& can it be trusted)?

If perimeter security is still something a user pursues, he/she should implement this with a trusted component that does exactly this perimeter security (such as a firewall), and not some python code trying to do it. If you have left behind the traditional perimeter security and want to apply some more modern zero trust approaches, you can do this as well. 
The existing security mechanisms that Home Assistant provides doesn't work in combination with Meshtastic, implementing new security mechanisms is kind of outside of the scope of this project as well. 

The risk of false security impression should not be underestimated. I'd rather err on the side of not protecting anything and leaving up to the user to protect it with their required & preferred measures based on their own security assessment. 

But what is definitely a big oversight on my part is that this security warning did not make it in the final version of the actual translation / text.

asjmcguire, 2025-04-29:
I'd say the best solution, would simply be - serve it on a different port, instead of the default Home Assistant web port. That's basically what other integrations do. If someone has their Home Assistant publicly accessible then it will be a port forward to 8123 or 443. So if the integration provides it's web service on a different port 8128 as an example, then someone would have to consciously port forward that to make it public

broglep, 2025-04-29:
> I'd say the best solution, would simply be - serve it on a different port, instead of the default Home Assistant web port. 

I wouldn't want to trust that if someone has his home assistant publicly available (without a firewall or reverse proxy), that only port 8123 or 443 is open

Running a separate HTTP(S?) server seems also like rather sub-optimal thing to do, both for those running home assistant as well as those who develop an integration (e.g. in terms of complexly/security/maintenance).

closed 2026-11-09 by jamesarich