meshtastic/home-assistant #63: Roadmap to official release

Sabering1, 2025-04-15:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/broglep/homeassistant-meshtastic/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

To install this project we need to use HACS

### Describe the solution you'd like

Native meshtastic integration, a roadmap would be good to have too.

### Describe alternatives you've considered

/

### Additional context

/

broglep, 2025-04-18:
As this project is getting more traction, we can start discussion about a roadmap. 

But I still foresee a long way until we could think about adding this integration to Home Assistant Core.
What is the issue with using HACS as a way of installation?

Sabering1, 2025-04-18:
Hard to find, HACS is also not the default for most default users (like me).
This integration is also not mentioned in the meshtastic docs, which makes it even harder to find.

https://meshtastic.org/docs/software/integrations/mqtt/home-assistant/

broglep, 2025-04-23:
This integration has now become more official & has been transferred to meshtastic github organisation. 
I think this also paves the way to eventually mention it in the official docs and add it to the HACS default repository, so that you don't have to setup a custom repo in HACS. But I think we will need to rely on HACS for some time to come

It would actually be great if HACS would be part of home assistant core, so that there is an easy way to step into many custom integrations

closed 2025-06-14 by garthvh