meshtastic/home-assistant #68: Feature request: Increase Node Limit

techyporcupine, 2025-04-28:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/meshtastic/home-assistant/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Some devices such as the RAK4631 with nRF52 chips can only hold a max of 80 nodes, which means some nodes just appear with their IDs in the logbook when the node hits the 80 node limit. The mobile apps and web UI will store nodes over the limit in their "storage." 

### Describe the solution you'd like

Is it possible to have this integration cache the node count over 80 (or 100)?

### Describe alternatives you've considered

Have an external list of node IDs and their names, but that would be a slight pain.

### Additional context

If this is hard to do, don't worry about it, but as this integration is the main way I communicate with the mesh, it has been somewhat problematic to not see node names all the time.

broglep, 2025-04-29:
Do you have notification platform enabled so that all meshtastic nodes are getting a notify entity in home assistant?

That would be the only location where currently would make sense to permanently store some information about a node as additional attributes (and fallback to that information for logbook)

It would be possible to build our own "node db", but home assistant does not really offer nice way to have custom persistent storage. In-memory cache would be another option but unclear if that makes sense. 

We also might need more than just name, essential whole node info, depending on the use case

techyporcupine, 2025-04-29:
I do not but will enable it now. Would that cause it to keep the names in memory?

I don't love this idea but can we store it in YAML file(s)?

broglep, 2025-04-29:
> I do not but will enable it now. Would that cause it to keep the names in memory?

No, but it enables the closest thing we have in this integration that resembles a node database. 
Currently only stores the node id and user id, but not long and short name. 

> I don't love this idea but can we store it in YAML file(s)?
Rather json to be in-line with how home assistant stores data, but yes, that is a workaround some integration use to store their own data, but nothing really nice to do for an integration.

If no persistence is required, rather prefer having it in-memory in such a case. Home assistant should usually be long running process, so in-memory could be sufficient. When having own persistence, there is also follow-up question how to clean up that database.

techyporcupine, 2025-04-29:
I do think in-memory would work pretty well for this use case.