meshtastic/home-assistant #76: Integration spews warnings in latest Home Assistant 2025.6.0

heyitsyang, 2025-06-13:
### Meshtastic Firmware Version

0.6.1

### Home Assistant System Health details

## System Information

version | core-2025.6.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.3
os_name | Linux
os_version | 6.12.23-haos
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4994
Installed Version | 2.0.5
Stage | running
Available Repositories | 2018
Downloaded Repositories | 23

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.05.5
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 938.2 GB
disk_used | 46.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.5.1), File editor (5.8.0), InfluxDB (5.0.2), Grafana (11.0.0), Network UPS Tools (0.14.1), ESPHome Device Builder (2025.5.2), Advanced SSH & Web Terminal (20.0.2), MariaDB (2.7.2), Mosquitto broker (6.5.1), Frigate (0.15.1), Zigbee2MQTT (2.4.0-1), Z-Wave JS UI (4.4.0), Nginx Proxy Manager (1.0.1), AWNET to HASS (1.1.3), Piper (1.5.4), Dynu DNS (1.0.1), Studio Code Server (5.19.3), Duck DNS (1.19.0), Whisper (2.6.0), openWakeWord (1.10.0)

</details>

<details><summary>keymaster</summary>

zwave_integration | zwave_js
-- | --
network_status | on

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 14
views | 18
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), eno1 (enabled, default, auto), docker0 (disabled), hassio (disabled), veth6df8239 (disabled), vethf617ab3 (disabled), vethbf9ed87 (disabled), veth1c06a8c (disabled), veth048e966 (disabled), veth66c2280 (disabled), vethca991c1 (disabled), veth068363b (disabled), veth211306c (disabled), veth0f10e1f (disabled), veth1bd472a (disabled), vethe3acf70 (disabled), veth12d169f (disabled), veth0fa99a3 (disabled), veth5112426 (disabled), veth353f2e2 (disabled), veth258a0de (disabled), vethb26a56a (disabled), veth03813d2 (disabled), vethc8ef930 (disabled), vethd54a05d (disabled), veth3971e8b (disabled), vethe6e6d14 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), eno1 (10.27.0.44/16), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), veth6df8239 (), vethf617ab3 (), vethbf9ed87 (), veth1c06a8c (), veth048e966 (), veth66c2280 (), vethca991c1 (), veth068363b (), veth211306c (), veth0f10e1f (), veth1bd472a (), vethe3acf70 (), veth12d169f (), veth0fa99a3 (), veth5112426 (), veth353f2e2 (), veth258a0de (), vethb26a56a (), veth03813d2 (), vethc8ef930 (), vethd54a05d (), veth3971e8b (), vethe6e6d14 ()
ipv6_addresses | lo (::1/128), eno1 (fe80::7418:35e:2283:b555/64), docker0 (fe80::c081:a1ff:fe73:e5d1/64), hassio (fe80::cc61:44ff:fe1a:39f2/64), veth6df8239 (fe80::702f:4dff:fe6c:e862/64), vethf617ab3 (fe80::f493:d1ff:fe08:4b72/64), vethbf9ed87 (fe80::b475:69ff:fe32:aff6/64), veth1c06a8c (fe80::94bd:deff:fe97:39a3/64), veth048e966 (fe80::483d:7aff:fe3b:12f8/64), veth66c2280 (fe80::445e:e9ff:fe79:b898/64), vethca991c1 (fe80::e829:60ff:fe9d:373b/64), veth068363b (fe80::e452:22ff:fe13:36d4/64), veth211306c (fe80::18d2:cdff:fea8:58c/64), veth0f10e1f (fe80::f066:ff:fe1b:7170/64), veth1bd472a (fe80::801e:95ff:fe53:439/64), vethe3acf70 (fe80::849f:56ff:fe49:da8d/64), veth12d169f (fe80::bcaa:abff:fe6a:340c/64), veth0fa99a3 (fe80::30e2:b7ff:fe51:2060/64), veth5112426 (fe80::94d3:25ff:fece:703/64), veth353f2e2 (fe80::34f3:2dff:fe77:49e0/64), veth258a0de (fe80::642c:fcff:fe7e:e666/64), vethb26a56a (fe80::6094:6ff:fe69:a8af/64), veth03813d2 (fe80::8cfe:83ff:febe:b476/64), vethc8ef930 (fe80::2c07:d5ff:feda:70e6/64), vethd54a05d (fe80::6092:c6ff:fe73:9ab6/64), veth3971e8b (fe80::447f:bbff:fee9:8512/64), vethe6e6d14 (fe80::c8ae:15ff:fef2:a99/64)
announce_addresses | 10.27.0.44, fe80::7418:35e:2283:b555

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 4, 2025 at 2:23 PM
-- | --
current_recorder_run | June 13, 2025 at 9:26 AM
estimated_db_size | 1721.44 MiB
database_engine | mysql
database_version | 10.11.6

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/meshtastic/home-assistant/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Version 2025.6.0 of home assistant spams (>10) warnings in the log:

This error originated from a custom integration.

Logger: py.warnings
Source: custom_components/meshtastic/aiomeshtastic/protobuf/localonly_pb2.py:12
integration: Meshtastic (documentation, issues)
First occurred: 9:26:21 AM (1 occurrence)
Last logged: 9:26:21 AM

/usr/local/lib/python3.13/site-packages/google/protobuf/runtime_version.py:98: UserWarning: Protobuf gencode version 5.28.2 is exactly one major version older than the runtime version 6.31.1 at meshtastic/aiomeshtastic/protobuf/localonly.proto. Please update the gencode to avoid compatibility violations in the next runtime release. warnings.warn(

### Reproduction steps

1. Just install HA 2025.6.0 and show the Home Assistant Core logs
2.
3.
...

### Home Assistant Debug logs

```text
N/A
```

broglep, 2025-07-02:
Spamming is maybe a bit exaggerated, but yes, I could verify that there are about 12 warnings being logged during startup. I also see such warnings for core integration in home assistant (e.g. apple tv)

we will have to live with those warnings for some time, as upgrading protoc  will on the other hand prevent running this integration with older home assistant versions that use older protobuf runtimes. I'd like to retain compatibility with at least 2024.12 which was using 5.28.3

jamesarich, 2026-11-09:
The deprecated calls are gone on main - a restart on 2026.10 logs no warnings from the integration. Reopen with the log lines if any come back on 0.7.0.

closed 2026-11-09 by jamesarich